Ultimaker/Cura #13478: Slicing a part, Cura 5.1.0 misses to slice a hole

JesConsuegra, 2022-10-05:
### Application Version

5.1.0

### Platform

Linux 64

### Printer

Hyoercube Evolution

### Reproduction steps

1. Load the part into Cura 5.1.0
2. Slice it
3. In preview mode, a hole has disappeared.

### Actual results

Printed the part, the hole is missing in the part.

### Expected results

That hole should have been made at slicing time. On other slicers (Prusaslicer, Superslicer, Ideamaker) that part slices properly
[missing-hole.zip](https://github.com/Ultimaker/Cura/files/9712750/missing-hole.zip)
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/9712800/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Easy to reproduce, just open the part and try to slice it.

Piezoid, 2022-10-05:
You have "Mesh Fixes" -> "Remove All Holes" enabled. This setting is meant to do exactly that: removes all holes in each slice. 
The horizontal slots and vertical holes are not affected because they split some slices in multiple parts rather than producing polygons/parts with vertical holes in them.

JesConsuegra, 2022-10-06:
Thanks!. But I can't see that "Remove All Holes" entry in "Mex Fixes", even if I set to see all controls, so I could not had set it.... And I cannot reset it.
[image]
[image]

JesConsuegra, 2022-10-06:
Thanks, This control  was somewhat hidden to me, so I could'nt have set it. Anyway I got to show all the mesh controls and there it is, solved!. Thanks again.

closed 2022-10-06 by JesConsuegra