Ultimaker/Cura #13479: When slicing a Dog Whistle "window," Cura calls for a floating inner wall.

Ergonomicmike, 2022-10-05:
### Application Version

5.2.0 beta

### Platform

Win 10 65-bit

### Printer

Ender 3 V2

### Reproduction steps

I am printing an (ultrasonic) Dog Whistle. I found an .stl here: https://www.thingiverse.com/thing:4970265.
I am using the Standard Quality profile (0.20 layer height) for the Ender 3 V2 with a 0.6 mm nozzle.

The Whistle is composed of three parts. Here's what the finished product looks like.

[image]

For this report, focus on the middle part with the curved/arched whistle "window,' Part B. (Download from the thingiverse link, above.)

Here's what Part B looks like from a distance in Cura:

[image]

Here's a zoomed in image of the window part after importing the slt into Blender. 

[image]

As you can see, the window is smooth with no strange irregularities. (TinkerCAD shows the same smooth arch.)

### Actual results

And now here is a zoom in of the window in Cura. Note the green inner wall hanging in mid-air.

[image]

### Expected results

Naturally, I didn't expect a wall to be floating in mid-air. 

I tried lower layer heights (e.g. 0.16, 0.12 mm). The problem becomes worse.

I suppose there might be some value in Cura that I can tweak to fix this. But I tried tweaking for a half hour without success.

I don't know - maybe Cura is doing what it's supposed to do and this part HAS to have a support?

(I printed it reversed in Z and it printed the window better, even with the bridge.)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Ultrasonic_Whistle_B for github.zip](https://github.com/Ultimaker/Cura/files/9713085/Ultrasonic_Whistle_B.for.github.zip)

MariMakes, 2022-10-05:
Hey @Ergonomicmike,

Thanks for the report 👍 

I see that the your flute contains a lot of what I would call wedges.
It's one of the shapes that the Variable Linewidth struggles with the most.

But there are settings that can help print these wedges better. 
[You can learn more about them here.](https://support.ultimaker.com/hc/en-us/articles/4792077687068-What-new-print-settings-are-introduced-to-Cura-5-) 

With just 2 profile changes I've been able to get this result.
Is this what you are looking for? 

[image]

If it is, do you agree it's not a cura issue and we can close the ticket?  👼

Ergonomicmike, 2022-10-05:
Perfect! I thought it might be a setting. Thanks!

Am closing now.

closed 2022-10-05 by Ergonomicmike

MariMakes, 2022-10-05:
[image]

Ergonomicmike, 2022-10-05:
BTW, the Dog Whistle works! I blew on it last night and the dog across the street woke up and looked at me to see what the "noise" was all about.