Ultimaker/Cura #13482: 1st base coat

fabnuss, 2022-10-05:
### Application Version

5 and 4.13 etc

### Platform

win 10

### Printer

alfawise u20

### Reproduction steps

Hello, the 1st base coat is always disgusting, I never have a clean and smooth bottom like on the top why?
Moreover, there is a bug on the way the lines are put, there are lots of holes between the lines ??? I made a skirt, it is perfect and the filling of the bottom is never perfect and does not make a smooth bottom why? please fix this bug or tell me what setting to do to avoid this! thank you

### Actual results

I already explained everything above

### Expected results

well if it was correct, the 1st base coat would be perfectly smooth like the one visible on the outside and it's really not the case

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

can't attach a file here???

GregValiant, 2022-10-06:
Thanks for the report.
I'm not familiar with your printer but I've never had an issue with initial layers on my printer.  Github requires that files be zipped before they can be posted.  Without a file to check we have nothing to go on.
Right click on the file in Explorer and select "Send To" and then "Compressed (Zip) folder" and post the zip folder here.  You can post a "bad" gcode file but if you load a model in Cura and set Cura up to slice and then use the "File | Save Project" command and then zip the 3mf file and post the zip file here that would be helpful as well.

GregValiant, 2022-10-06:
Attachments don't come through when you email.

fabnuss, 2022-10-06:
I did not receive any error message, sorry. I resent the files to you again.
tell me if it doesn't work, I'll send them to you separately

Fabian

 

Envoyé: jeudi 6 octobre 2022 à 17:03

GregValiant, 2022-10-06:
Do this...
Right click on the file you want to post.
Select "Send To...".
Select "Compressed (ZIP) Folder". 
Post the Zip folder here.

fabnuss, 2022-10-06:
it's good that's what I did, the file is zipped.
why haven't you received it yet?

 
 

Envoyé: jeudi 6 octobre 2022 à 19:41

MariMakes, 2022-10-07:
Hey @fabnuss ,

Welcome to the Ultimaker Github 🎉 

Can you try this for me?
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

fabnuss, 2022-10-07:
Bonjour, Écoutez j'ai déjà envoyé 2x tous les fichiers zip sous divers formats. Je vais pas faire que ça. Je ne vois pas où sur github il faut joindre les fichiers c'est pour cela que je les ai envoyé par email et apparemment votre serveur n'accepte pas où je ne sais pas ce qui se passe. Il n'y a que chez vous qu'il y a des problèmes de réception car j'ai jamais de soucis avec mes amis. Je veux bien encore le faire une fois mais c'est la dernière. Faite moi un print écran pour me dire ou joindre le fichier svp. Cordialement Fabian --Envoyé à partir d’un Smartphone Android avec GMX Mail.

MariMakes, 2022-10-10:
Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.

I'm sorry you are having so much trouble uploading the file we need to help you troubleshoot.
We really want to help you to reach the quality of 3D print you are looking for but for that we need to learn what is going wrong.

It would help if you could share pictures of what is going wrong. 
Sharing your project file would be just as valuable.  

Instead of screenshots, I made a recording of where to click to upload your project to github.
https://user-images.githubusercontent.com/40423138/194861352-0da90102-171b-412e-9850-31a1009ce88f.mp4

Good luck!

fabnuss, 2022-10-10:
[AU20_ship_4.zip](https://github.com/Ultimaker/Cura/files/9746908/AU20_ship_4.zip)
[image]
here I hope that with that you can tell me what's wrong! for your info version 5 of cura is slower than version 4.13 I did the calculation for the same room and 4.13 gave almost 1 hour less, it's huge. Moreover the beta versions I'm always wary of it, it's still not finished versions. in addition to the 4.13, the way I set it up, it comes out almost perfect, in fine 0.1mm mode it's beautiful, smooth, no flaws. it's just the bottom that's a concern. Here again, thank you for your help on github, because nothing indicated that you had to click on the text because I was looking for a button or to see that the cursor changes to a hand to indicate that it is the link to attach file! This github is really fucked up lol. See you soon. Fabian

MariMakes, 2022-10-10:
Hey @fabnuss,

Thank youuuu 😄 
Your picture was super helpful! 💪 

There seems to be something off with your leveling. 
I think there is too much distance between your nozzle and your buildplate:
[image]

Can you doublecheck the leveling procecdure form you ALfawise u20?
I think that is where you can make the most difference. 

Usually we cannot learn a lot about your settings from a gcode, but I took a look anyways. 
The only setting that seems a bit off is this one: expand_skins_expand_distance = 1.5

Can you let us know if you are getting better results after leveling your buildplate? 
Because it doesn't seem to be a Cura issue.

fabnuss, 2022-10-11:
> Hello,
> thank you for responding to me so quickly. here is my project and what you asked me.
> my printer the alfawise u20 and identical to the Longer3d brand.
> thank you and see you soon
> Fabian

Envoyé: jeudi 6 octobre 2022 à 12:29

fabnuss, 2022-10-11:
Hello,
thank you for responding to me so quickly. here is my project and what you asked me.
my printer the alfawise u20 and identical to the Longer3d brand.
thank you and see you soon
Fabian

 
 

Envoyé: jeudi 6 octobre 2022 à 06:06

MariMakes, 2022-10-11:
Hey @fabnuss, 

Can you try this leveling tutorial on your printer?
https://www.youtube.com/watch?v=naXbW_9VLQU
I believe something is wrong with your bedleveling.

fabnuss, 2022-10-11:
ok i will look into that, thanksHave a good dayFabian--Envoyé à partir d’un Smartphone Android avec GMX Mail.

fabnuss, 2022-10-11:
hi Mari,

 

another small question, what is the best solution to put supports on a part to be printed? is it the zigzags or the tree structure, it's to get the best rendering, because on normal media there are always too many small bits to clean up and when that doesn't tear the impression.
thank you for your reply.

I received my new glass top, it is very flat, I will install it tomorrow and redo all the settings :-)

Envoyé: mardi 11 octobre 2022 à 11:07

 

https://www.youtube.com/watch?v=naXbW_9VLQU

MariMakes, 2022-10-25:
Hey @fabnuss,

Since this is a hardware issue and not something that we can resolve in Cura, I'll be closing this issue.
I hope you understand 🙏 
Good luck with your 3D printing adventures.

closed 2022-10-25 by MariMakes