Ultimaker/Cura #13483: Slicing Bug, Cura is asking to be reported here

nissaba, 2022-10-05:
### Application Version

5.1.0

### Platform

Mac OS 12.6

### Printer

Anycubic I3 Mega S

### Reproduction steps

use plugin to auto-orientation optimal speed then slice

### Actual results

got an erre asking to report this bug. 

### Expected results

to get the object sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[AI3M_Drone_parts.3mf.zip](https://github.com/Ultimaker/Cura/files/9717850/AI3M_Drone_parts.3mf.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/9717855/cura.log.zip)

GregValiant, 2022-10-06:
Thanks for the report.
Cura 5.1 has a bug on Windows installations but this might be the first report of it happening on a Mac computer.  The bug seems to have been fixed in 5.2 for Windows.
If you Group the models and then move and rotate the group on the build plate you may very well get it to slice.
Someone from the Cura team will take a look at this.
I rotated the group 90° and it sliced.
[image]

nissaba, 2022-10-08:
> Thanks for the report.
> Cura 5.1 has a bug on Windows installations but this might be the first report of it happening on a Mac computer. The bug seems to have been fixed in 5.2 for Windows.
> If you [...]

Yes I did move the parts after opening the issue, and it did slice. My guess is that the item where place in a way that it can't arrive to a solution. I normally let Cura place the items on the place using r-click arrange all models. For this I used the extensions to get the optimum orientation then got it to auto rearrange the items, that failed then I moved things a bit and this worked.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker