Ultimaker/Cura #13485: Error while slicing the model

Jayang21, 2022-10-05:
### Application Version

5.2.0 - beta 

### Platform

Windows 10 

### Printer

Ultimaker 2 extended +

### Reproduction steps

I am getting errors while slicing a 3d model of a tesla cyber truck 
i tried different prints but it still gives me the error 

### Actual results

It wont slice the model 

### Expected results

it should be slicing the model since i just used it an hour ago 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

63601206 on thingverse

GregValiant, 2022-10-06:
Well, I get the confetti page when I try to find that model.  Without a project file and Cura log file there really can't be an investigation.

MariMakes, 2022-10-06:
Hey @Jayang21,

Welcome to the Ultimaker Cura Github 🚀 

Do you have a project file for us?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 😄

MariMakes, 2022-10-06:
Hey @Jayang21,

Thanks for getting back to us. 
What you shared is the .zip file that you downloaded from Thingiverse.

I'm looking for a project file. The file contains the settings that you use for slicing in Cura and ends with .3mf. 
If I just slice your STL I get a successful result. 

[image]

[UM2E_Tesla_Cybertruck.zip](https://github.com/Ultimaker/Cura/files/9725115/UM2E_Tesla_Cybertruck.zip)

Do you have a project file for us?
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2022-10-21 by github-actions[bot]