Ultimaker/Cura #13490: Cura Crash during Slice

muckaluck1, 2022-10-06:
### Application Version

5.1.0

### Platform

Windows 10 64 bit

### Printer

Ender 3 V2

### Reproduction steps

Load STL and slice with profile used to slice other STLs in the same set.

### Actual results

Cura crashed.

### Expected results

STL should slice
[CE3PRO_WWA_SCRPT_LV2_B.zip](https://github.com/Ultimaker/Cura/files/9725362/CE3PRO_WWA_SCRPT_LV2_B.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9725370/cura.log)

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Toot.

Piezoid, 2022-10-06:
Could you try the 5.2.0-beta? It was able to slice your model (at least on my linux machine).

GregValiant, 2022-10-06:
@piezoid - I can't get the 3mf file to slice using 5.2 on my Windows machine.

Exporting the model as an STL the file is 114mb.  After simplifying the model in 3D Builder it is 5.5mb and the smaller file did slice.  3DBuilder didn't find any errors but the image below is the view in MeshMixer.  There are a couple of errors but that center pillar is bad and there is something going on all along the base of the wall on the left.  Mesh Mixer was able to make repairs but they don't look good as those areas with errors simply disappeared (next image).
[image]

[image]

I'm not happy with Mesh Mixer.  "Erase" is not the same as "Repair".

muckaluck1, 2022-10-06:
> Could you try the 5.2.0-beta? It was able to slice your model (at least on my linux machine).

I'll def give it a shot but if you got it to slice any chance you could send it over?

Piezoid, 2022-10-06:
@GregValiant Interesting, I was building and slicing with the `5.2.0-beta` tag. This might be the windows specific bug that was reported before.

There is a slight gap where the side pillars split into 4. And if this gap were closed, there would be face overlaps: the mesh parts are nor properly merged. When sliced as is, it leaves unsupported layers: 
[image]

GregValiant, 2022-10-06:
@Piezoid I noticed that.  Since they aren't properly part of the main model maybe Mesh Mixer decided they weren't necessary.  Still, that violates the law that "The software shall not adlib."

_"This might be the windows specific bug that was reported before."_  Man I hope not.  I'm really sick of that one.  It's one thing to be the greeter here when there is a chance to learn something.  Endless repetition is simply boring.

Piezoid, 2022-10-06:
> Could you try the 5.2.0-beta? It was able to slice your model (at least on my linux machine).

Do you mean the sliced .gcode? I'd rather not send gcode, you'll loose the feedback loop for tunning the print and it is a bad habit to print gcode sliced by others.

Could you first try to slice this simplified version of the STL:  [WWA_SCRPT_LV2_B_fixed.zip](https://github.com/Ultimaker/Cura/files/9727011/WWA_SCRPT_LV2_B_fixed.zip)
(I used blender and a kitchen sink sequence of operators: Delete Loose, repeated Degenerate Dissolve until a fixpoint is reached, weld modifier, 95% decimation, Delete Loose again)

Piezoid, 2022-10-06:
> Thanks for the file, unfortunately it also crashes. I should have looked at all the files before I started this!
@muckaluck1 Sorry about that, I don't even have a Windows machine to try what I sent you...

Anyway, here the gcode, hopefully getting your project unstuck: [CE3PRO_WWA_SCRPT_LV2_B.zip](https://github.com/Ultimaker/Cura/files/9727417/CE3PRO_WWA_SCRPT_LV2_B.zip)

Sliced with 2.5.0-beta on linux. Be sure to load it in cura for inspection before sending it! I took the liberty of enabling Thin Walls to support the side pillars.

@GregValiant I run CuraEngine built with the undefined and address sanitizer. This should have caught problems that can be silenced on one platform but making it crash on other, but it didn't detected any issue at runtime. It could be that the crash happens in one of the libraries (clipperlib) which are not compiled with the sanitizers. 
I don't know what was the chain of events that lead to the crash in #12707, but it might be that clipperlib is crashing on these empty polygons...

muckaluck1, 2022-10-06:
Looks good!  I will throw this on the plate after I get done with this next section.  I appreciate all the help.  The pillar areas where the gaps are is for attaching doors.  It's a beautiful print, happy to be able to continue!

GregValiant, 2022-10-06:
Here is the bottom of the column that disappeared.
[image]

And it looks hollow for at least a little ways.  The column to the right is no prize either.
[image]

Piezoid, 2022-10-06:
As muckaluck1 said, these are built-in supports. I didn't realized that. So, after all, this is sort of the expected behavior.

GregValiant, 2022-10-07:
I used Mesh Tools and split the original model into several pieces.  I threw them all away except for the base structure and that sliced (after a bit of time) in 5.2beta.
I'm inclined to agree with you regarding Expected Behavior.  The model is put together poorly.

Il leave this as a bug report as I'm not quite sure what else to do with it.