Ultimaker/Cura #13491: Slicing failed

smnbgn, 2022-10-06:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Anycubic Predator

### Reproduction steps

1. Slicing the file mercury_south_1_2_10_7.stl file from [here](https://www.thingiverse.com/thing:3503256) **at 75% scale** fails

### Actual results

Slicing failed error message is displayed

### Expected results

G-code output

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Slicing files with the file

[AP_mercury_south_1_2_10_7.zip](https://github.com/Ultimaker/Cura/files/9726413/AP_mercury_south_1_2_10_7.zip)

[cura.zip](https://github.com/Ultimaker/Cura/files/9726446/cura.zip)

GregValiant, 2022-10-06:
Thanks for the report.
I couldn't slice the file in 5.2beta.  Although the model didn't show any errors, Mesh Tools in Cura said there were some errors so I uploaded it to a [Repair Site](https://formware.co/OnlineStlRepair).  It didn't find any errors but it did add vertices and triangles.  The repaired file is larger as well.
-> Vertex count changed from 115957 to 123652 (+7695)
-> Triangle count changed from 232202 to 248646 (+16444)

The repaired model did slice.

[image]

I'll leave this as a bug and a slicing error and someone from the Cura Team will take a look at it.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker