Ultimaker/Cura #13492: slice error

jerry1935, 2022-10-06:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anet et4 pro

### Reproduction steps

Slices thingiverse # 4709670 gargyle pattern at 50mm size
Repeated at 100mm size.

### Actual results

successful at 50 mm, error at 100mm.  Error is repeatable OLNY if you slice at another size first.

### Expected results

pattern sliced for saving

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-10-07:
Hey @jerry1935 

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 

Do you have a project file for us? It contains the printer and settings I need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 😄

jerry1935, 2022-10-07:
This may be a memory issue (leakage).  My machine has 64gb of DDR4-3200.  Today Cura did the 100 mm size ok and that said it couldn't slice a 75 mm image                                                                                               
[AET4PRO_Gargoyle_Updated_Support_Free_icz.zip](https://github.com/Ultimaker/Cura/files/9734909/AET4PRO_Gargoyle_Updated_Support_Free_icz.zip)

closed 2022-10-07 by jerry1935

GregValiant, 2022-10-07:
I'm on an older laptop with 8gb of memory and running Cura 5.2beta and it sliced fine at 100 and at 75mm tall.  So I'm unable to duplicate the problem.
There was an often reported bug in 5.1 for Windows that would prevent slicing.  A workaround was to move and/or rotate the model on the Cura build plate.  That would often allow a model to slice.
I would advise moving to Cura 5.2beta.  From what I have seen so far that bug has been fixed.