Ultimaker/Cura #13493: ERROR Impossible to cut

dumser2, 2022-10-06:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Anet A6

### Reproduction steps

I added a model for printing, I had already printed it before.

### Actual results

Impossible to cut, ERROR

### Expected results

What to do?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Есть файл

MariMakes, 2022-10-07:
Hey @dumser2,

Welcome to the Ultimaker Cura Github 🚀 

Do you have a project file for us? It contains the printer and settings I need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 👼

dumser2, 2022-10-08:
[truba.zip](https://github.com/Ultimaker/Cura/files/9739921/truba.zip)
[image]

MariMakes, 2022-10-10:
Hey @dumser2 ,

Thanks for sharing your model.
I've been able to slice your model in Cura 5.1 and 5.2 Beta.
[image]

But from your screenshot I can see that have non latin characters as your translations. 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/9644
This is supposed to be resolved in https://github.com/Ultimaker/Cura/pull/13394, and will be shipped with Cura 5.2
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/9644

closed 2022-10-10 by MariMakes

dumser2, 2022-10-10:
[image]

MariMakes, 2022-10-11:
Hey @dumser2,

This is supposed to be resolved in the 5.2 Stable that will be published soon.

If you are in a hurry, a workaround could be to change your path of the temp folder:

Search for environment variables in the windows search bar.
This will open system properties and there you can click on Environment Variables.
Edit the TEMP variable.
Change the Variable value to something like C:\Temp