Ultimaker/Cura #13495: New Tree Support Trunk Diameter setting not adjusting collision volumes

Ghostkeeper, 2022-10-07:
### Application Version

5.2 beta

### Platform

Ubuntu 22.04

### Printer

Ultimaker 3

### Reproduction steps

1. Load a rather tall model that needs support at the top, nearby the rest of the model.
2. Enable support and select tree support.
3. Reduce the Tree Support Trunk Diameter setting such that it limits the width of the support branches.
4. Let it slice and observe layer view.

See also project file below.

### Actual results

The support branch curves away from the model as if it continues to get wider, but in actuality it doesn't get wider due to the new Tree Support Trunk Diameter setting.

### Expected results

* The branch should not try to increase the distance from the model.
* Using the Tree Support Trunk Diameter setting I should be able to reduce the size of the print on my build plate more (as written in the setting description).
* Using the Tree Support Trunk Diameter setting, the support branches should be able to pass through more narrow openings of the model (as the new diameter would allow).

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

This constructed example shows that the tree support branches try to avoid the object, but due to the Tree Support Trunk Diameter setting are considered wider than they are really printed. As a result, the branch goes further and further away from the model unnecessarily, which compromises the reliability of the print.
[image]

The implementation of this setting only adjusted the way the circles are drawn for at the end ([here](https://github.com/Ultimaker/CuraEngine/blob/bb4c436022199a042757872503faa1a37d3a6830/src/TreeSupport.cpp#L85)), not the actually simulated size of the circles around the nodes. So the tree branches behave as if they are wider than they really are.

I suggest that instead of the current implementation, you make the following changes:

* In https://github.com/Ultimaker/CuraEngine/blob/bb4c436022199a042757872503faa1a37d3a6830/src/TreeSupport.cpp#L317, limit the diameter to the maximum diameter provided by the setting.
* Same here where the code is duplicated: https://github.com/Ultimaker/CuraEngine/blob/bb4c436022199a042757872503faa1a37d3a6830/src/TreeSupport.cpp#L379
* Same here where the code is also duplicated: https://github.com/Ultimaker/CuraEngine/blob/bb4c436022199a042757872503faa1a37d3a6830/src/TreeSupport.cpp#L428
* Remove the current implementation in the draw function. If the node gets the correct (limited) size, the draw function should just draw that size without any changes.

Here is a project file that reproduces the problem:
[branch_max_diameter_weirdness.zip](https://github.com/Ultimaker/Cura/files/9735521/branch_max_diameter_weirdness.zip)

MariMakes, 2022-10-10:
Hey @Ghostkeeper,

Thanks for your report 👍 

I can reproduce your issue on my side. 
[image]

That indeed looks like undesired behavior. 
To me it looks more like a featuregap than a bug, but I'll bring it up to the team to take a look at. 

Fingers crossed 🤞

MariMakes, 2022-10-19:
Hey @Ghostkeeper,

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.

I discussed your issue with the team and we believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself. 

I hope you understand 👼

DerGenaue, 2022-10-20:
[image]

Ghostkeeper is the top 1 contributor to the Cura repo and top 2 contributor to the CuraEngine repo...

Ghostkeeper, 2022-10-23:
That doesn't mean though that the issues I find are top 1 priority ;)

I tried to make it easy to fix by investigating a bit where the issue arose, but fixing and testing it still takes time, and it's not the most important issue in the world.

DerGenaue, 2022-10-23:
yeah; but "Our community is a big part of that by requesting new features and reporting bugs." and "you or anyone else is more than welcome to work on this issue and create a pull request yourself." feels a bit funny and out of place in that context 😄

MariMakes, 2022-10-24:
We are really trying to improve handling community requests. We are looking at ways to free up time from our in-house developers to spend more time on reducing the waiting time for PRs. 😅 

I saw the tree support fork and it really sparked my interest. 🤩 
If you or they have any questions they can reach out to me and I'm happy to address any concerns or have the team take a first look at it.

DerGenaue, 2023-07-04:
Can be closed now that the new Tree Support algorithm hast been merged into Cura 5.4

closed 2023-07-11 by MariMakes