Ultimaker/Cura #13499: Simple-geometry vase mode bottoms don't bond to vase section

richfelker, 2022-10-07:
### Application Version

5.x

### Platform

Any

### Printer

Any

### Reproduction steps

1. Make a large cube.
2. Slice it for printing in vase mode
3. Print

### Actual results

At least the first-side wall breaks off easily from the base. This is because, in the transition layer to vase mode, Cura only chooses flow at polygon segment granularity. The entire first wall that's supposed to have increasing thickness starting from near zero is extruded at a single flow (not sure whether it's the average or the minimum flow). As a result, the part after the nozzle moves up slightly from the previous layer is not making any contact to bond with the previous layer.

### Expected results

Segments of the transition layer should be finely subdivided to give accurate thickness-based flow all the way along. Also, the transition layer should probably start at something like 25% of the layer thickness rather than 0, because very low flow does not reliably extrude on most printers.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

GregValiant, 2022-10-07:
Thanks for the report.
One "lap" around a model in Vase mode is an increase in the Z of 1 layer Height.  A cube having 4 sides then has 4 Z moves in one lap.
This is the Gcode of the first spiralized layer of a 50mm cube sliced in Vase mode.  Layer Height was 0.20mm and Line Width was 0.60mm and a 0.40 nozzle.
G0 F300 X130.487 Y132 **Z1** ;move up to working height of layer 4
G0 F9000 X134.03 Y134.03 ;combing
G1 F1500 E993.28479 ;combing
G0 F9000 X134.7 Y134.7 ;combing
;TIME_ELAPSED:812.705700
;LAYER:4
;TYPE:WALL-OUTER
;MESH:X-Y Ref Block.stl
G1 F2040 X85.3 Y134.7 E995.74936
G1 X85.3 Y85.3 E998.21394
G1 X134.7 Y85.3 E1000.67852
G1 X134.7 Y134.7 E1003.14309
G1 X85.3 Y134.7 **Z1.05** E1005.60767
G1 X85.3 Y85.3 **Z1.1** E1008.07225
G1 X134.7 Y85.3 **Z1.15** E1010.53682
G1 X134.7 Y134.7 **Z1.2** E1013.0014

This is the flow analysis of those last 8 lines:
Delta E         mm/sec            mm³              Ext Len    Time            mm³/sec      Act Flo%
2.46458		34	           5.92801	49.40008  1.45294            4.08        100.00%
2.46458		34	           5.92801	49.40008	1.45294	         4.08	100.00%
2.46458		34	           5.92801	49.40008	1.45294	         4.08	100.00%
2.46457		34	           5.92799	49.39988	1.45294	         4.08	100.00%
2.46458		34	           5.92801	49.40008	1.45294	         4.08	100.00%
2.46458		34	           5.92801	49.40008	1.45294	         4.08	100.00%
2.46457		34	           5.92799	49.39988	1.45294	         4.08	100.00%
2.46458		34	           5.92801	49.40008	1.45294	         4.08	100.00%

In the Gcode portion above you see that the nozzle does 2 laps.  The first is at 1.0mm and puts down a .2mm thick layer and the second starts to step up .05mm per line of extrusion.  But the Flow Rate remains the same as it would be for a full .20 layer height.  This over-extrusion generally leaves a small ridge on a model and has been a source of complaint for some people.  The flow of the spiral section should really be 25%, 50%,, 75% and finally at 100% to match the change in _"height above the previous layer"_ rather than strictly by "Layer Height".  The @smartavionics fork of Cura includes that sort of behavior.

So if your first layers are breaking off I think you need to look elsewhere as Cura over-extrudes the first 3 steps of the first spiral layer in Vase Mode (that is of a cube and other shapes would have more steps).
I do not like printing vases with silky PLA's because of the poor layer adhesion.  Also, you will note that I did the slice at .6 Line Width and a .4 nozzle.  The wider line width helps a lot with strength and durability.  That is important when printing aircraft and watercraft parts.

If you have different Flow numbers that show "polygon segment granularity flow" please post them.  At this point I see I don't see a bug but rather Expected Behavior.

richfelker, 2022-10-08:
OK, I'll need to make a better test case and post it here I think. The one that failed today had long straight sides but small rounded corners that obscure what's going on in the gcode, but I think I can give a brief explanation of what I'm seeing right away:

- Layers 0-2 are bottom (0.2, 0.4, 0.6).
- Layer 3 is single wall at 0.8, then repeated as full flow vase with Z increasing from 0.85 to 1.0.
- Layer 4 is where things get bonkers. It starts at 1.049 with under half flow for the first side, then increasing up to full flow by the 4th side.
- Layers 5 and up are all normal vase with full flow.

It looks like layer 4 (the 5th layer and the 6th loop) is where things are failing. I don't get why it's suddenly using reduced flow at this stage where it's always a full 0.2 mm from the previous layer.

richfelker, 2022-10-08:
Hmm... could this be a bug where it's inconsistent about use of `bottom_layers` vs `initial_bottom_layers` in the spiralize code? I have 3 of the latter but 5 of the former, which matches up exactly with where the problem is happening... And indeed I see some `bottom_layers` under `if (spiralize)` in `WallsComputation.cpp`...

richfelker, 2022-10-08:
OK, it's not that one because changing it didn't fix it, but setting `bottom_layers` to 3 does fix it. So I guess there's another wrong one somewhere...

richfelker, 2022-10-08:
Looks like the wrong one that's breaking it is in `FffGcodeWriter::processSpiralizedWall`. There are at least a few other places where layer numbers are compared to `bottom_layers` too. Clearly these are all wrong.

richfelker, 2022-10-08:
Confirmed fixed with those all changed to `initial_bottom_layers`.

```
diff --git a/src/FffGcodeWriter.cpp b/src/FffGcodeWriter.cpp
index 214ecad49..167badf5d 100644
--- a/src/FffGcodeWriter.cpp
+++ b/src/FffGcodeWriter.cpp
@@ -2075,7 +2075,7 @@ void FffGcodeWriter::processSpiralizedWall(const SliceDataStorage& storage, Laye
             last_seam_vertex_idx = storage.spiralize_seam_vertex_indices[layer_nr - 1];
         }
     }
-    const bool is_bottom_layer = (layer_nr == mesh.settings.get<LayerIndex>("bottom_layers"));
+    const bool is_bottom_layer = (layer_nr == mesh.settings.get<LayerIndex>("initial_bottom_layers"));
     const bool is_top_layer = ((size_t)layer_nr == (storage.spiralize_wall_outlines.size() - 1) || storage.spiralize_wall_outlines[layer_nr + 1] == nullptr);
     const int seam_vertex_idx = storage.spiralize_seam_vertex_indices[layer_nr]; // use pre-computed seam vertex index for current layer
     // output a wall slice that is interpolated between the last and current walls
@@ -2931,7 +2931,7 @@ bool FffGcodeWriter::processSupportInfill(const SliceDataStorage& storage, Layer
                 if ((! wall_toolpaths.empty() || ! support_polygons.empty() || ! support_lines.empty()))
                 {
                     int layer_nr = gcode_layer.getLayerNr();
-                    if (layer_nr > (int)infill_extruder.settings.get<size_t>("bottom_layers"))
+                    if (layer_nr > (int)infill_extruder.settings.get<size_t>("initial_bottom_layers"))
                     {
                         // bit of subtlety here... support is being used on a spiralized model and to ensure the travel move from the end of the last spiral
                         // to the start of the support does not go through the model we have to tell the slicer what the current location of the nozzle is
diff --git a/src/WallsComputation.cpp b/src/WallsComputation.cpp
index 2dbea8f18..5c106aadb 100644
--- a/src/WallsComputation.cpp
+++ b/src/WallsComputation.cpp
@@ -59,7 +59,7 @@ void WallsComputation::generateWalls(SliceLayerPart* part)
                 settings.get<bool>("support_enable") && !settings.get<bool>("fill_outline_gaps");
 
         generateSpiralInsets(part, line_width_0, wall_0_inset, recompute_outline_based_on_outer_wall);
-        if (layer_nr <= static_cast<LayerIndex>(settings.get<size_t>("bottom_layers")))
+        if (layer_nr <= static_cast<LayerIndex>(settings.get<size_t>("initial_bottom_layers")))
         {
             WallToolPaths wall_tool_paths(part->outline, line_width_0, line_width_x, wall_count, wall_0_inset, settings);
             part->wall_toolpaths = wall_tool_paths.getToolPaths();
```

GregValiant, 2022-10-08:
Post a gcode file.  If the flows are off then my Excel macro should find it.  I'm not able to reproduce this when I alter the Initial Bottom Layers.  I still get 99.98% to 100.01% flow in all layers.

richfelker, 2022-10-08:
Without and with patch applied:

[cube28-vase-bug.gcode.txt](https://github.com/Ultimaker/Cura/files/9738115/cube28-vase-bug.gcode.txt)
[cube28-vase-good.gcode.txt](https://github.com/Ultimaker/Cura/files/9738116/cube28-vase-good.gcode.txt)

Diff is trivial:

```
--- cube28-vase-bug.gcode	2022-10-08 01:25:22.310743830 +0000
+++ cube28-vase-good.gcode	2022-10-08 01:21:17.742559794 +0000
@@ -1081,9 +1081,9 @@
 ;LAYER:4
 ;TYPE:WALL-OUTER
 ;MESH:cube28.stl
-G1 F6624 X103.7 Y131.3 Z1.05 E0.40162
-G1 X103.7 Y103.7 Z1.1 E0.57374
-G1 X131.3 Y103.7 Z1.15 E0.74586
+G1 F6624 X103.7 Y131.3 Z1.05 E0.91798
+G1 X103.7 Y103.7 Z1.1 E0.91798
+G1 X131.3 Y103.7 Z1.15 E0.91798
 G1 X131.3 Y131.3 Z1.2 E0.91798
 ;TIME_ELAPSED:48.207889
 ;LAYER:5
```

GregValiant, 2022-10-08:
The extrusion rate for those three lines is 43.75%, 62.5%, and 81.25%.  All other lines in the file are at 100%.
That is not actually the first layer of spiralize.  Layer 3 is the start layer.  This snippet is from your "bug" gcode file:
;LAYER:3
;MESH:cube28.stl
G0 X131.3 Y131.3 Z0.8
M204 S6000
;TYPE:WALL-OUTER
G1 F10800 X103.7 Y131.3 E0.91798
G1 X103.7 Y103.7 E0.91798
G1 X131.3 Y103.7 E0.91798
G1 X131.3 Y131.3 E0.91798
**G1 X103.7 Y131.3 Z0.85 E0.91798
G1 X103.7 Y103.7 Z0.9 E0.91798
G1 X131.3 Y103.7 Z0.95 E0.91798
G1 X131.3 Y131.3 Z1 E0.91798**
;TIME_ELAPSED:47.143843
;LAYER:4
;TYPE:WALL-OUTER
;MESH:cube28.stl
G1 F6624 X103.7 Y131.3 **Z1.05 E0.40162****
G1 X103.7 Y103.7 **Z1.1 E0.57374**
G1 X131.3 Y103.7 **Z1.15 E0.74586**
G1 X131.3 Y131.3 **Z1.2 E0.91798**

Exactly what version of UltiMaker Cura are you using?
I ask because I don't get those extrusion numbers with 5.2beta and interestingly enough, those are exactly the same percentage numbers that occur with the SmartAvionics Master version (although his version puts those flow numbers on the layer where spiralize actually starts).
Posting a project file would be good.

smartavionics, 2022-10-08:
> Confirmed fixed with those all changed to initial_bottom_layers.

Yes, those changes are required although only the first diff is really relevant to this bug (I think).

In my fork, I have the first correct already but the other two I need to change, thanks for reporting.

richfelker, 2022-10-08:
> That is not actually the first layer of spiralize. Layer 3 is the start layer.

Exactly. This is because it's looking at the wrong setting to assess whether this is the first layer of spiralize.

> Exactly what version of UltiMaker Cura are you using?
> I ask because I don't get those extrusion numbers with 5.2beta and interestingly enough, those are exactly the same percentage numbers that [...]

I'm using CuraEngine branched from git mainline, https://github.com/Ultimaker/CuraEngine/commit/d30d09852999, with various local fixes (unrelated) applied, and don't have a setup of the Cura frontend handy. Hitting the bug depends on having `bottom_layers` set to something greater than `initial_bottom_layers` and should happen however you're invoking CuraEngine (directly or via UI) as long as you have a version new enough for `initial_bottom_layers` to be a separate setting.

The analysis how the bug happens should be clear. `spiralizeWallSlice` reduces flow based on the `is_bottom_layer` boolean that `FffGcodeWriter::processSpiralizedWall` computed using the wrong number of bottom layers.

rburema, 2022-11-09:
Solved by the PR provided by the original poster of this issue!

closed 2022-11-09 by rburema