Ultimaker/Cura #13500: failed to slice unknown error

Robandhooch, 2022-10-08:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

Loaded file
selected slice

### Actual results

Error message that unknown error occurred, failed to slice
[image]
[AV_LVL3_FrontB.zip](https://github.com/Ultimaker/Cura/files/9738072/AV_LVL3_FrontB.zip)

### Expected results

should have sliced the file and then i could save it

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[AV_LVL3_FrontB.zip](https://github.com/Ultimaker/Cura/files/9738075/AV_LVL3_FrontB.zip)

Robandhooch, 2022-10-08:
[AV_LVL3_FrontB.zip](https://github.com/Ultimaker/Cura/files/9738076/AV_LVL3_FrontB.zip)
This keeps happening randomly.  up till a couple weeks ago have never seen this error.  Here's the project

MariMakes, 2022-10-10:
Hey @Robandhooch,

Thanks for your report 😊 

Did you try to slice it on Cura 5.2 Beta? It sliced succesfully for me. 
[image]

We were able to fix a number of causes of crashes that were introduced in Cura 5.0.|

Can you let me know if you are still stuck after trying Cura 5.2 Beta? 👼 
You can download it here: https://github.com/Ultimaker/Cura/releases

Robandhooch, 2022-10-10:
So funny enough, changing layer height to .21 it slices fine.  very odd, as
i've been slicing this set of files just like i always do at layer height
.20 without issue.

MariMakes, 2022-10-11:
Hey @Robandhooch.

I have to be honest, I don't understand your last comment. 

Are you still stuck? 
Or can I close this issue? 👼

Robandhooch, 2022-10-11:
If i leave all things the way my slicer settings is on the file fails to
slice.  if i increase the layer height to .21, from .20, it will slice
without error.
So yes there is still an issue, but i have no idea where to go from here
with it.

MariMakes, 2022-10-12:
Hey @Robandhooch,

Even if I change the layer height, I can slice your project file on Cura 5.2 Beta 
[image]

Perhaps you can share your logs and we can find an error of what is going wrong...
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

Robandhooch, 2022-10-12:
I do not know how to do logs, and if the new version doesn’t have this problem I guess the simples solution is to just update cura. Will I have to reload my profiles, or will it automatically draw them in? Sent from Mail for Windows From: MariMakesSent: Wednesday, October 12, 2022 4:25 AMTo: Ultimaker/CuraCc: Robandhooch; MentionSubject: Re: [Ultimaker/Cura] failed to slice unknown error (Issue #13500) Hey @Robandhooch,Even if I change the layer height, I can slice your project file on Cura 5.2 BetaPerhaps you can share your logs and we can find an error of what is going wrong...You can find your log file here:Windows: %APPDATA%\cura<Cura version>\cura.log or usually C:\Users\\AppData\Roaming\cura<Cura version>\cura.log