Ultimaker/Cura #13503: Sandwiched walls :-)

hogglash, 2022-10-08:
### Is your feature request related to a problem?

Yes, i want to build some parts light but strong

### Describe the solution you'd like

Hi. Im building as a newbie 3d printed rc planes. Im not the guy that makes the cad files.
I thinking of a nice feature ii would like.
If i have a barreld shaped section of a rc plane then it would have been nice to just "klick" make double wall from a single and chose the infill betwwen those layers.
Lets say i make a extra wall inside 15 mm apart with 10% infill.
That would be extremely nice.
H / SWE

### Describe alternatives you've considered

-

### Affected users and/or printers

All

### Additional information & file uploads

[image]

MariMakes, 2022-11-01:
Hey @hogglash,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I believe that this is something you can achieve by adjusting the correct settings. 
Personally, I would change this to achieve the same result: 
[image]

Would you still like to keep this issue open or do you agree that we can close it?

hogglash, 2022-11-01:
Hi, thats for the answer.
But not really what i was thinking about.
Im thinking that if i have a one or two layered outer wall that i by chosing something can make a new inner wall lets say 10-15 mm from the outer wall and if i chose to use infill inbetween those. Like a sandwiched wall

MariMakes, 2022-11-01:
Hey, 

This reminds me of the hollow features in for example MeshMixer.
https://formlabs.com/eu/blog/how-to-hollow-out-3d-models/

If you use that feature you can create STLs that look like this with a few clicks
[image]

Is this what you mean?