Ultimaker/Cura #13505: Pause at Layer + Print one at a time = Layer Shift

Malte-G, 2022-10-08:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Anycubic i3 Mega S

### Reproduction steps

1.Load 2 Objects and select Print one at a time. 
2.Slice
3. Add Postprocessing: Pause at Height(Layer)
4. Select desired Layer of the object that will be printed second.
5. Slice Again
6. Print.

### Actual results

The print will go as usual but the pause during printing of the second object will occur at a higher layer than selected in Postprocessing.

The difference between desired layer and actual layer appears to be precisely the total number of layers of the first object.

I suspect that Cura counts layers differently in the preview than it does during printing.

### Expected results

The Pause should occur at the selected layer regardless of number of previously printed objects.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura Bugreport.zip](https://github.com/Ultimaker/Cura/files/9740203/Cura.Bugreport.zip)

GregValiant, 2022-10-09:
Thanks for the report.
Because Pause at Height is a post-processor it runs after the gcode has been put together so yes, the layer numbering changes.  The preview layer numbers are only good for the initial preview.
When printing "one at a time" each model sits on LAYER:0 and so if you have 10 models on the build plate and each is 100 layers tall there will be 10 LAYER"0's.  The preview would show the top layer as LAYER:1000 but the reality is that you have 10 prints that are each 100 layers tall.  If you want to pause 1/2 way up the last print and enter layer 950 as the pause layer it won't work as there won't be a layer 950 in the gcode file.  The pause would need to be at the 10th LAYER:50.
You can copy and paste the Pause code in the gcode to the place you want it.  Look closely and you can find the "Last XY", "Resume Z", and "Resume E" locations. Plug those numbers into the code you pasted.
Another oddity happens when you open one of those "one at a time" gcode files in Cura.  It will show the layers as they are in the gcode with all parts starting together on LAYER:0 when they really print one at a time.
I don't see a bug here.  Confusing?  Yes.  Bug?  No.  It's Expected Behavior because Pause at Height runs as a post-processor.
@marimakes might have a comment on whether this should remain open.  It is something that has come up before.

MariMakes, 2022-10-10:
hey @Malte-G,

Welcome to the Ultimaker Cura Github 🚀 

@GregValiant is correct, what you are seeing is by design. 
I've changed the issue from Bug to Feature status since it's more of a feature request.

Can we have Pause at Layer when Printing One At a Time?
I'll bring it up with the team to discuss. 🤞

MariMakes, 2022-10-26:
Hey @Malte-G, 

I brought up your suggestion to the team. 

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

I hope you understand 🙏

Malte-G, 2022-10-26:
Hi @MariMakes,

thank you for the response and for your great work on Ultimaker I greatly appreciate your efforts.

Malte

GregValiant, 2023-10-27:
I had forgotten about this one.
I have a pull request in for a collection of small post processors I call "Little Utilities".  One of them is to renumber the layers of a "One at a Time" file (where each model starts at layer:0) to "All at Once" numbering (consecutive layer numbers).  The main benefit is that if Pause at Height runs after renumbering you can print each model a different color, or add a nut to each model at the same point (that would require multiple instances of Pause at Height).  A secondary effect is that the gcode will preview correctly in Cura.  There is an option to "Revert" the numbering by adding a second instance of Little Utilities.

Here is a beta of the script.  You can unzip the folder and put "LittleUtilies.py" into the "scripts" folder within your configuration folder.
[LittleUtilities.zip](https://github.com/Ultimaker/Cura/files/13188074/LittleUtilities.zip)

GregValiant, 2026-10-01:
Starting with 5.14.0, PauseAtHeight can be configured to pause at the Cura Preview Layer in a single model when in One_at_a_Time mode.
I'll close this.

closed 2026-10-01 by GregValiant