Ultimaker/Cura #13507: Is the Flow Rate correctly calculated by the Cura Engine/Ender 3 v2 Profile?

Ergonomicmike, 2022-10-09:
### Application Version

5.0 - 5.2.0 beta

### Platform

Win10 64-bit

### Printer

Ender 3 v2

### Reproduction steps

Not really an Issue, per se. More like something to ponder.
-------
Admittedly still a newbie (4 months now with a new Ender 3 v2). But hopefully learning quickly.

When I first received my Ender, I learned about the "Calculate your E-steps" procedure. And so, using my Display to manually "move" 1000 mm of filament through the extruder, I found that I had to reset my Extruder steps from the stock 93 for the Ender 3 v2 to 101.2 steps to move a certain amount of filament. That's about a 9% increase. (And, of course, this setting has nothing to do with Cura. It's a hardware thing.)

(The stock 93 steps/mm comes from most everyone's firmware (in config.h) for the Ender 3 v2. It could be that that number is wrong for later Enders.)

After that, I thought that everything should be okay. Except that I was over-extruding in Cura, as observed from experimentation (printing Calibration Cubes and such). 

For the nozzles and temperatures that I use, I had to lower my flow rate in Cura to 93%. (For PLA. 95% for PETG since PETG is more viscous and didn't over-extrude as much as PLA.)

Then, a month after this, it occurred to me that the REAL test for extrusion/flow rate should be mass. (As Stephan of CRC Kitchen uses when [he does maximum flow rate tests](https://www.youtube.com/watch?v=UNJdv5bFGOg).)

That is, if 1000 mm of white PLA filament from the spool weights 3 grams, then if you extrude (through your hot nozzle) 1000 mm of white PLA, then the blob of filament that results should weigh 3 grams. (Assuming that you are within the extrusion limits of your hot end (that is, you are extruding slowly).)

### Actual results

Well, guess what? I was over extruding. By 6%. Which is almost the same percentage that I had increased my E-steps earlier!

So I might be causing my own problem here by "correctly" setting my E-steps, but which in turn, caused over-extrusion?

I have since lowered my Flow Rate (on a per Material basis) in Cura. For example, I use 93% for PLA. (Although I'm tempted to simply set my Extruder step rate back to 93 and leave flow in Cura at 100%.)

So now all this has me wondering if Cura is properly extruding/flowing filament on a mass basis? I ask because I occasionally  see posts in forums that say that Cura is known for over-extruding. (Admittedly old posts.)

And if there is a mis-calculation, is it in the slicing part of Cura or if it's in the Machine Profile part? That is, I wonder if the Ender 3 v2 Profile is correct?

### Expected results

I'm not smart enough to know how to model something in Cura to have a certain mass (say, 3 grams) as a Sanity Check to see if Cura itself is commanding the correct extrusion/flow of 1000 mm of filament in a model. (Such a model would have to take into account layer height, layer widths, etc. - which I presume Cura already does.)

Can someone test this? Perhaps a model for the Calibration Extension guys to ponder?

I know there is the model of the topless cube with no infill, where we're supposed to measure the thickness of the walls as a test for over/under-extrusion. But I think that a test of weight would be better/more direct measure of how the slicing model is commanding extrusion. 

And even when Cura is commanding the correct amounts of filament (by mass), things like printing at too low a temperature, printing with a partially clogged nozzle, printing too fast can all cause under-extrusion. So a Calibration model for mass might be helpful.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Moderators, if this would be better posted in Requests for Enhancements, feel free to move this.

Asterchades, 2022-10-09:
Cura can only set its E values based on what you tell it under ideal circumstances. It has no choice but to assume your E steps are properly set at the printer, that the filament diameter matches exactly with what you've told it, and that you aren't exceeding the physical capabilities of your printer. As soon as one of those things is no longer true, it has no way of knowing or compensating for it, such as explicitly changing the flow modifier.

Consider a single extrusion on the edge of a standard 20mm calibration cube. At 0.4mm wide, 0.2mm tall, in theory this will take roughly 1.6mm² of filament (it will be slightly less due to the rounded edges but work with me here). If your filament is perfectly round and exactly 1.75mm in diameter, this will take 665.2um of filament being fed, which Cura will round to 0.665mm to satisfy Marlin.

Thing is, all of this depends on things being precise. If your filament is not perfectly circular it may require more or less to be fed through. If the diameter varies from exactly 1.75mm it will require more or less to be fed. And this then depends on your E step value being set properly, perfect filament grip, _and_ the steppers landing exactly where they're supposed to.

All of this is why Cura will show a filament mass down by the time estimate. If you've told it the appropriate density for your filament, this shows what it should weigh at the end - which I believe isn't affected by the flow percentage you've set, since that's for calibration.

If I'm to be perfectly honest, it sounds like you might have an issue at your extruder. That you had to increase the steps seems like your feed gear is wearing, reducing its effective diameter. This not only results in the E-steps needing to be increased but is also going to affect your extruder's grip on the filament - perhaps causing you to have inadvertently raised your step value higher than it needed to be.

Ergonomicmike, 2022-10-09:
Thanks for your thoughts.  I don't think it's an issue with my (metal) Extruder per se because 1) my Extruder (Ender printer) is virtually brand new, 2) there is no debris around the gear and, 3) I just took it apart yesterday. No visible wear on the gear.

I've tugged on the filament while feeding. Seems tight. I've increased the physical pressure on the gear. No change. (Although Stefan points out that too much pressure deforms the filament - although this shouldn't change the mass of the filament.) And I've seen videos from other Ender owners where they have to increase their steps.

And if my Extruder were skipping, and I was increasing steps to compensate for that, then how could I be over-extruding? Wouldn't the failure mode be to under-extrude?

Regardless, none of this matters because I have is set now so that when 1000 mm of filament is commanded, that's what feeds through the Extruder. As a Sanity Test, it even feeds 1000 mm when pushing the filament through the Hot End. (Indicating that the nozzle is not bottle necking the Extruder, causing it to skip,) So, on the assumption that Cura calls for 1000 mm of filament, that's what my setup delivers out the nozzle.

Yes, it could be that my filament has a larger diameter than spec. I don't have an accurate way of measuring that. But since all the different filaments I've tried (brands/types) over-extrude, it seems that filament diameter is not the main problem.

You're correct that it's difficult to calculate how much a squished first layer should weigh per linear unit. While we might know the layer height (assuming that commanding a 0.20 mm layer results in a 0.20 mm squish - but Z-offset would affect this), I'm not so sure that layer width can be assumed. For if Cura doesn't calculate the flow correctly, then the layer will be too wide or too narrow from ideal.

And then there's the shape of the squished filament, which probably takes some interesting math to model/integrate.

Maybe the way to do a Calibration "model" for mass is to extrude many long diagonal lines floating a few millimeters in the air. (Instead of on the Plate.) This way we can be sure that the filament is round, extruded with no resistance, and will have the cross section of the nozzle diameter making the math very easy. A few long lines minimizes having to do the math at reversals.

GregValiant, 2022-10-09:
Cura does the calculation by volume as the mass can be affected by water absorption of the filament, by manufacturer, the material used for colorant, and other factors that are near impossible to measure.
The actual measured diameter of the PLA filament I use is 1.72mm.  If I was to leave Cura at 1.75mm diameter then I would under-extrude by ~4%.
The formula is "Volume of Filament In" = "Volume of Extrusion Out".  When the ratio is 1:1 you are at 100% flow rate.
The Volume of Filament is "(PI)r² x FilamentLength" and the Volume of Extrusion is LineWidth x LayerHeight x Length.  Because Cura varies the line width in the 5.x versions it isn't really possible to backtrace the E numbers that are in a Gcode file.  An E value may seem to be less than it should be but that could be because Cura narrowed the line width for that extrusion.

Calibrating the E-steps is necessary on most Creality printers and their clones.  The boards come with a default of 93.  My e-steps are at 97.  The steps of the XYZ motors essentially never need calibration.

Never use a single or double wall "calibration" cube to determine flow.  All you will accomplish is to be able to print single wall models with walls of an exact thickness.  Models that are like that and require precise single wall thicknesses don't come along very often.  If you calibrate your E-steps to that sort of model, ALL your other types of prints WILL be under-extruded.

Skins are the best evidence of over/under extruding.  A microscope is the best tool to inspect with but a decent magnifying glass will work.  Once your e-steps are calibrated (and it sounds like they are) then let the model talk to you.  Your eyeballs will tell you more than playing with numbers will.  If there are gaps between the extrusions and you can see the layer below then you are under-extruding.  If there are ridges and smears at the end of an extrusion then you are over-extruding (or printing too hot - different topic).

closed 2022-10-10 by Ergonomicmike