Ultimaker/Cura #13509: Bluescreen

Golf3gti1, 2022-10-09:
### Application Version

5.1.0

### Platform

Win 10 

### Printer

Artillery Genius 

### Reproduction steps

When I start Cura and insert an object I get a bluescreen with the message " IRQL_NOT_LESS_OR_EQUAL".  
But this only happens with Cura. Can someone help me there what 

### Actual results

Computer collects and restarts the computer New 

### Expected results

Das Problem habe ich erst seit ich 5.1.1 installier hatte. aber habe das 5.1.1 deinstalliert aber dennoch habe ich das probloem weiter hin 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Unfortunately I do not have a log file

MariMakes, 2022-10-13:
Hey @Golf3gti1,

I got confirmation from our support experts that you were able to resolve this issue.

For other people encountering the issue: It was the GPU driver, it needed to be updated. A good website to look through and resolve it can be found here: https://www.tomshardware.com/how-to/fix-irql-not-less-or-equal-errors-windows

I'll close this issue since it's been resolved.

Golf3gti1, 2022-11-05:
Hello 

Unfortunately, it has not gebaruch. 
It only happens when I start Cura, put an object in and edit it. With other programs I have no problem. All drivers are up to date