Ultimaker/Cura #13510: Support generated in the air

vovodroid, 2022-10-09:
### Application Version

5.1.1

### Platform

Win10 x64

### Printer

Biqu B1 SE ABL

### Reproduction steps

Slice attached project, see what happens:

[image]

[image]

### Actual results

Support tree is cut in the air

### Expected results

Tree is growing from the ground ))

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project:
[BB1ABL_RC-Body.zip](https://github.com/Ultimaker/Cura/files/9741940/BB1ABL_RC-Body.zip)

GregValiant, 2022-10-10:
Move the model away from the edge of the build plate.  Maybe there isn't enough room to put the support down where Cura wants to put it.
An alternative is to select your model, click on Per Model settings and then Select Settings.  Go down to the last setting in the Support Section which is called "Drop Down Support Mesh".  It should be enabled. by default.
[image]

vovodroid, 2022-10-10:
It's not matter of placement. Sometimes it works, sometimes not. Changing forth and back any option and reslicing could help, as well as reloading. Also sometimes two-roots tree generated:

[image]

Would like me to enable logs? I also can build Cura if needed.

GregValiant, 2022-10-10:
There was a bug that allowed Tree support to grow outside the build area.  I wonder if this is related to the fix for that(?).
Someone from the team will take a look.
If you could post a Cura.log file that would help. If you were able to point to a section of code you believe is suspect then that would help as well.

vovodroid, 2022-10-10:
> If you could post a Cura.log file

Currently it's working. I'll upload log if I encounter it again.

> if you were able to point to a section of code you

I'm not familiar with Cura code, just can build it if necessary (so didn't try).

DerGenaue, 2023-07-04:
FYI: This probably has been solved in the new Cura 5.4 release