Ultimaker/Cura #13514: If you zoom out too much, the view gets stuck and can't zoom back in.

staffert, 2022-10-10:
### Application Version

5.20 beta

### Platform

windows 10

### Printer

Creality CR10S Pro V2

### Reproduction steps

1. Zoom way out.
2. Try to zoom back in.
3. 

### Actual results

can't zoom back in.

### Expected results

should be able to zoom back.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

    none.

MariMakes, 2022-11-08:
Hey @staffert,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We've been getting a number of reports on this issue, so I'll bring it up tot the team to see if we can improve this behavior.
Fingers crossed 🤞

Distillah2, 2022-11-08:
It seems that when Cura is opened with admin credentials, the issue is not there. 
It seems that the issue has something to do with rights. Hopefully this is a hint for the solution.

CWempe, 2022-12-03:
I just tried Cura v5.2.1 (used v4.x before) and experienced this issue.

Opening Cura as Admin did not help for me.

I even cannot change the camera angle by clicking the cube-icons in the loft bottom corner.

Is there a workaround to control the camera again?
Otherwise it seems I need to downgrade again. :(

TimeRider62, 2022-12-15:
I have downgraded to Cura v5.0.0, this was the last version that the mouse responded correctly.  It is just that simple. The last version that it worked in was 5.0.  I have searched high and low but can not find an answer.
Edition Windows 11 Pro
Version 22H2
OS build 22621.963

TimeRider62, 2023-01-01:
Still broken and still no answers.

MatCastel, 2023-01-02:
I had te same problem. Be sure to have the plugin "camera tool" in marketplace setting (the "gear" icon"). I've accidentally disabled it and had no pan, zoom, rotation 
[image]

TimeRider62, 2023-01-02:
Well, son of a gun! That was it! I had no idea it was disabled or even an option! Thank you!!!

Distillah2, 2023-01-03:
The Camera tool is not available in the manage packages environment (the gear tool). 
It is not disabled, it is just not there.
Where can I find it for installtion?

TimeRider62, 2023-01-03:
I found it here.

[image]

MariMakes, 2023-01-03:
Hey @Distillah2,

It should be bundled with your installation.
So I definitely expect it on the list!

Just in case: the plugin lives in this folder (on Windows)  \share\uranium\plugins\Tools
And you can download the files that you need here: https://github.com/Ultimaker/Uranium/tree/main/plugins/Tools

CWempe, 2023-04-03:
Worked for me, too. 👍

No idea how this plugin got disabled.

But I guess this issue can be closed.
Unless Cura is shipped with this plugin disabled, wich would be an issue that should be fixed.