Ultimaker/Cura #13515: slicing failure

mizzat907, 2022-10-10:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Lotmaxx Shark

### Reproduction steps

I loaded the file.
centered the object and adjusted infill.
hit slice
Then received the error "slicing failed"

### Actual results

Cura goes back to allowing me to slice the object if i want. I have tried the beta version which is 5.2 as well and got the same error. 

### Expected results

I expected it to slice the print and allow me to save to my computer or micro SD card. 
[LS_double-swipe.zip](https://github.com/Ultimaker/Cura/files/9743232/LS_double-swipe.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

attached is the project file

GregValiant, 2022-10-10:
Thanks for the report and the Project FIle.
I checked the model in Cura using the Mesh Tools plugin.
_"The model is not watertight and may not print correctly and consists of 41 sub-meshes"._  It's those curved pieces around the perimeter that seem to be a problem.
I was able to get the original model to slice in 5.2beta by changing the layer height to 0.20mm.

Here is the view in Mesh Mixer.  The magenta areas indicate errors in the model.  (This is the coolest looking set of errors I've seen.)
[image]

I exported the model as an STL and uploaded it to this [OnLine Repair](https://formware.co/OnlineStlRepair) site.  This is the report.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 120 Non-manifold edges (?)
--> 120 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 167765ms ------
-> Vertex count changed from 22984 to 81722 (+58738)
-> Triangle count changed from 45964 to 170674 (+124710)

I think the repaired model would have sliced at 0.10 layer height but it was taking too long and I got bored.  Canceling the slice of the .1 layer height attempt caused Cura to crash.
I'll leave this as a bug and someone from the Cura team will take a look.  @Piezoid you may want to take a look at this one as well.  I've attached my project file with the repaired model.
(**FYI** there would be about 36,600 retractions in a 0.10 layer height gcode file with the print time around 25 hours.)
[GV_double-swipe_repaired_model.zip](https://github.com/Ultimaker/Cura/files/9745598/GV_double-swipe_repaired_model.zip)

[image]

Piezoid, 2022-10-10:
I'll have a look at your fixed model. (indeed, it might be the stitching of polygons broken by vertex merging)
I have also fixed the model by doing a boolean union with all the non connected pieces (there were many of them): [LS_double-swipe_r.zip](https://github.com/Ultimaker/Cura/files/9746646/LS_double-swipe_r.zip)
I can slice it with 5.2 beta.

GregValiant, 2022-10-10:
@Piezoid was your slice good at .10 layer height?

GregValiant, 2022-10-10:
I tried repairing but indeed the Union operation seems to have fixed it.  It will slice at 0.10 layer height in 5.2b.

@mizzat907 did you generate this model yourself?  Best practice is to merge all parts into a single solid before exporting from CAD.  In this case the errors all relate to the separate pieces around the periphery of the middle part.
The gcode file is 110mb and 3.5million lines of code.  There were 17,000 retractions and a print time of 24 hours.

mizzat907, 2022-10-12:
Thank you for your help. I did not generate this mode myself I found it on printables.com. Sent from my iPhone

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker