Ultimaker/Cura #13517: Slicing stays in 'Slicing...' state without any progress

Triffen, 2022-10-10:
### Application Version

5.2.0 Beta (using modern appimage)

### Platform

Manjaro Linux, KDE Plasma

### Printer

Ender 3 V2

### Reproduction steps

1. Open Cura 5.2 Beta
2. Drop any stl inside
3. Click 'Slice'

### Actual results

It says 'Slicing...' without progress. If I press "Cancel" or delete the model, it says "Slicing failed" and recommends to report a bug. If I try to slice again, it shows "Unable to slice".

### Expected results

I expect it to slice.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Freshly opened Ultimaker-Cura-5.2.0-beta-linux-modern.AppImage with STL dragged and dropped inside and 'Slice' clicked:
[image]

After clicking 'Cancel':
[image]

'Unable to slice' state:
[image]

Project file:
[filament clip miniUnableToSliceCura5-2beta.zip](https://github.com/Ultimaker/Cura/files/9745521/filament.clip.miniUnableToSliceCura5-2beta.zip)

GregValiant, 2022-10-10:
Thanks for the report and the project file.
There was a bug in the Windows version of Cura 5.1 that prevented certain models from slicing unless they were moved on the build plate and/or rotated.  It seems that the bug fix for Windows may have transferred that bug to Mac and Linux.  Your project sliced quickly on my Windows system.
Someone from the Cura team will take a look.  (This appears to be a duplicate of #13483.)

Triffen, 2022-10-10:
Thanks!
I have tried to move and rotate the model but it's all the same.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker