Ultimaker/Cura #13521: "Zoom toward mouse direction" won't stay checked

adamsusernameis, 2022-10-10:
### Application Version

5.2.0-beta

### Platform

macOS Monterey 12.5.1

### Printer

Creality CR-6 SE

### Reproduction steps

1. Open preferences menu
2. Activate "Zoom toward mouse direction"
3. Close preferences menu
4. Open preferences menu

[image]
[image]
[image]
[image]

_Note: It also unchecks upon clicking a different tab without closing the preferences menu
Example:
1. Open preferences menu
2. Activate "Zoom toward mouse direction"
3. Click "Materials" tab in preferences
4. Return to "General" tab
Result: "Zoom toward mouse direction" is disabled_

### Actual results

"Zoom toward mouse direction" is unchecked/disabled.

### Expected results

1. Scrolling via center mouse button zooms window in and out
2. Holding right mouse button + dragging orbits window
3. Holding SHIFT + right mouse button + dragging pans window

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Issue remains after restarting/reinstalling version.
Have had this issue with the previous version 5's as well; been using 4.13.1 instead.

MariMakes, 2022-10-11:
Hey @adamsusernameis,

Thanks for the report. 👍 
It took a couple of tries but I've found someone able to recreate your issue (I'm on windows myself).

Somehow it seems to be a combination of Invert the Direction of the Camera Zoom and Zoom Towards Mouse Direction. 
Good find!

I'll bring it up to the team to see what we can do. 
Fingers crossed 🤞

adamsusernameis, 2022-10-11:
That's great! Thank you @MariMakes !

MariMakes, 2022-10-19:
Hey @adamsusernameis 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9771

Thanks for the report! 👍

Ohmarinus, 2022-10-23:
I wanted to add that I have this problem with multiple camera settings. They do not always stay checked, sometimes I have to restart Cura three times in order for the changes to 'stick'.

Windows 10, Cura 5.2.1

adamsusernameis, 2023-07-04:
@MariMakes 

Been a while, and since creating this report (CURA-9771), I've just kept using 4.13.1

Just tried to download 5.4.0 - wouldn't open because program "can't be scanned for malicious content"
Tried 5.4.0 beta1 - opened, but wouldn't load model
Tried 5.3.1 - opened, loaded model, but the issue with being unable to zoom/pan with the mouse remains
Tried 5.2.2 - opened, loaded model, still the issue with mouse zoom/pan

It's quite frustrating, as I'd really like to update the program. My MacOS is current & I really have no idea why this is happening.

I'm using a mac magic mouse..does Cura no longer support it?
Thanks

etyrnal, 2024-06-08:
for several versions, on Ubuntu, the option to enable the zoom toward mouse direction is greyed out.  can't be selected.  used to work in cura 4.x.x i think it work initially in 5.x.x also.  but then just went away.  beyond annoying.