Ultimaker/Cura #13522: Can't slice my file. Not very computer savy.

ToddyWestside, 2022-10-10:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

JGAurora A5S

### Reproduction steps

I downloaded a STL file off of Hero Forge and the slicer won't cut it. I downloaded it with another file and the other one worked fine. 

### Actual results

Nothing same thing. 

### Expected results

I want to print my mini. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

none

MariMakes, 2022-10-11:
Hey @ToddyWestside,

Welcome to the Ultimaker Cura Github 🚀 
I love slicing hero forge mini's, and I am happy to help out. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

ToddyWestside, 2022-10-12:
Hi Mari.

I figured out that the supports I had chosen were wrong. Once I changed
them Cura worked fine.

Thank you for getting back to me.

MariMakes, 2022-10-12:
Hey @ToddyWestside,

Thanks for the update. 😊 
I'll close your issue. 

Happy printing 👍

closed 2022-10-12 by MariMakes