Ultimaker/Cura #13523: 3MF file reading problem

urbas80, 2022-10-11:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

custom

### Reproduction steps

Hello, when I open a file in 3MF format in Cura, the objects are not arranged correctly, i.e. differently as in the program from which I exported the objects. Below I am attaching files and screenshots.

[bug_array.zip](https://github.com/Ultimaker/Cura/files/9751137/bug_array.zip)
[image]
[image]

### Actual results

Objects arrange badly on the buildplate.

### Expected results

I want the same arrangement of objects as in the program from which I export objects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[bug_array.zip](https://github.com/Ultimaker/Cura/files/9751155/bug_array.zip)

MariMakes, 2022-10-11:
Hey @urbas80,

Welcome to the Ultimaker Cura Github 🚀 
That's a super interesting report. 👍 
There is something odd going on here. 

I can reproduce it on my side. 
[image]

Even if I uncheck the Ensure Models are Kept Apart and Automatically Drop Modles To Buildplate they seem moved. 
[image]

I'll bring it up to the team to take a look. 
Fingers crossed 🤞

urbas80, 2022-10-11:
I will add that with one model and a cube inside, the same thing happens
[image]
[bug1model.zip](https://github.com/Ultimaker/Cura/files/9754109/bug1model.zip)

MariMakes, 2022-10-11:
What is the software that you are using to design/export? 🤔

urbas80, 2022-10-13:
I'm use Rhino 3D 7

MariMakes, 2022-10-17:
Hey @urbas80,

Thanks again for the report.
I've discussed your issue with the team and we've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9755

Thanks for the report! 👍 

I also think that https://github.com/Ultimaker/Cura/issues/13387 might be related 🤔