Ultimaker/Cura #13528: Strange line width vs. speed behavour

Theannor, 2022-10-11:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

I have two strange things happen with a very simple model

### 1. Change of speed and line width where it should not be needed

With this very simple OpenSCAD Model a can reproduce a behaviour of a more complex box layout

a = 50 ;
b = 100 ;
translate([-a,0,0]) cube([b, 0.8, 20], center=false);
translate([0,-a,0]) cube([0.8, b, 20], center=false);

Rendering to an stl and slicing this leads to ( I will add the test_x.stl for this example an add it as a zipfile)
)
[image]

Where the nozzle slows down when it comes to the center of the cross like structure. At the same time the line width increases as the speed goes down due to the constant flow. 

Changing the above model sligthly

a = 0 ;
b = 100 ;
translate([-a,0,0]) cube([b, 0.8, 20], center=false);
translate([0,-a,0]) cube([0.8, b, 20], center=false);

 ( I will add the test_l.stl for this example an add it as a zipfile)

[image]

The second one now keeps the speed, the line width and the flow at a constant rate. 

That second one also produces a faster (10% for this settings) and cleaner print than the first one.  

Playing around with "Flow equalization ration" does not change the behave

### 2. Single lines behave different

Using again the above cross example

a = 50 ;
b = 100 ;
translate([-a,0,0]) cube([b, 0.8, 20], center=false);
translate([0,-a,0]) cube([0.8, b, 20], center=false);

Slicing it again there is most of the time a line of this
[image]

within the result. Constant speed, width and flow, as I would expect for all lines

This line is somehow randomly located, there are sometime two of them. Changing for example the layer height and reverting it back again and reslice does lead to different lines. Sometimes you need to rotate to the backside but the line is present.

### Actual results

The a constant flow-rate for a straight "block" can be achieved by keeping the speed and the line width fixed. There is no obvious reason to change the speed and the width

### Expected results

For this kind of object I would expect the flow rate to be kept constant by keeping the speed and the line width konstant

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here are the stl-files for the test
[test3.zip](https://github.com/Ultimaker/Cura/files/9757746/test3.zip)

GregValiant, 2022-10-13:
I sliced your model and found a surprising number of line segments on each layer.  If I were to hand code it there would be 12 lines per layer.  Cura decided Layer:10 needed 180 lines.  It appears that each line is broken into multiple segments so each may have it's own speed and E value.

I read the gcode into AutoCad.  The images below are the plan view and all 100 layers are there.
This is the left left leg of the model.  The model is in white and the extrusion lines are in red.  This is how it should look.  The model is straight and square and the extrusions lines are straight and square.
[image]

This is the right leg.  Note that the end moves are no longer square to the model.  There are three separate paths used for the short segment on the end.  They are all wrong.
[image]

This is the center section.  You can see the multiple paths coming from the left (in the yellow shape).
The dimension points to another problem.  That entire extrusion run on every layer is at an angle.  The dimension should be 0.20 as it is at the bottom.  You can see that the dimension extension line does not line up with the extrusion line coming from the top.
[image]

At any rate, after confusing myself and getting side-tracked I did check the flow (0.20 Layer Height and 0.40 Line Width)
Max flow rate:  3.2mm³/sec or 145.8%
Avg flow rate:   2.7mm³/sec or 123.7%
Min flow rate:  2.4mm³/sec or 103.4%

Theannor, 2022-10-17:
This does correspond with the behavior of the print itself. 
So is this a wrong setting that I could correct myself?
Or is it somehow a bug?

MariMakes, 2022-11-08:
Hey @Theannor,

Sorry, it took us a while to get back to you 😞 

Your issue reminds me of https://github.com/Ultimaker/Cura/issues/13395

> In Cura 5.1 we improved the toolpath simplification algorithm. The algorithm that reduces the resolution of the print has been rewritten, providing a more constant resolution to allow for greater precision without causing buffer underruns. I've noticed that if you alter your extrusion Area Deviation to a very low value, you'll have the behavior back that you are looking for.

We've added a ticket to the backlog with the intent to improve this behavior for super simple single walled models. 

Would you agree it's a duplicate of https://github.com/Ultimaker/Cura/issues/13395 and we can close this issue?

Theannor, 2022-11-08:
Hello @MariMakes

I agree with you, https://github.com/Ultimaker/Cura/issues/13395 sounds very similar to my observed problem and also the images look similar. 

Duplicate ist okay for me. I have subscribet the other issue, so I will try to test if there ist a fix available

Thanks for the answer and the help

closed 2022-11-08 by MariMakes