Ultimaker/Cura #13530: Tree Support - BASE ONLY - Adds zigzag support to other overhangs that branches don't connect to from baseplate support

tymax6, 2022-10-11:
### Application Version

5.1.1

### Platform

win 10

### Printer

ender 3 v2

### Reproduction steps

1. add model with multiple over hangs that the branch from build plate won't reach.
2. Tree supports > from build plate only
3. slice

### Actual results

look under anything that would otherwise be supported if 'everywhere' was supported and you will find it added zigzag support under overhang.

### Expected results

no zigzag support under overhangs that don't have a tree leading up to it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3 HC_XYJoyner_Long_lm10luu Generic PETG lw 0.5mm lh 0.3mm if 20% ext1 225C bed 75C.3mf.zip](https://github.com/Ultimaker/Cura/files/9759181/CE3.HC_XYJoyner_Long_lm10luu.Generic.PETG.lw.0.5mm.lh.0.3mm.if.20.ext1.225C.bed.75C.3mf.zip)

GregValiant, 2022-10-12:
Thanks for the report and the Project File.
I can reproduce this in 5.2beta as shown below:
[image]

Support Interface is being added where it should not be.
Someone from the Cura team will take a look.

MariMakes, 2022-10-17:
Hey @TJoverkill,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 

One of our developers already discovered this during the Cura 5.2 Beta. 
I just took a look at the latest build and it seems to have been resolved there. 
So if you download Cura 5.2, once it is available, you can find a fix there. 

[image]

closed 2022-10-17 by MariMakes