Ultimaker/Cura #13533: Seemingly random M104 S20 in gcode

aaron-casper, 2022-10-11:
### Application Version

5.1

### Platform

Windows 10

### Printer

Voxelab Aquila

### Reproduction steps

Sliced a STL to print, attempted to print.

Partway through laying down raft layers, my printer (Voxelab Aquila/Ender clone) began failing to extrude.  The display indicated a target nozzle temp of 20C.

Checked GCODE manually, found M104 S20 inserted between translations.

Manually removed erroneous M104 instruction, print succeeded

### Actual results

GCODE generated includes erroneous M104 S20 instruction at line 225 in included file.

[axe_hanger_broken.zip](https://github.com/Ultimaker/Cura/files/9760031/axe_hanger_broken.zip)

### Expected results

[cura.log](https://github.com/Ultimaker/Cura/files/9760040/cura.log)
[axe_hanger_broken.zip](https://github.com/Ultimaker/Cura/files/9760041/axe_hanger_broken.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project file in .zip with GCODE including erroneous instruction.

Included cura.log

GregValiant, 2022-10-11:
Thanks for the report.
I've never seen that one before.  It should have been M104 S200.
Someone from the Cura Team will take a look.
In the meantime look around on the floor around your printer and see if you spot an extra zero laying around.

aaron-casper, 2022-10-12:
Hi Greg,

Thanks for following up so quickly.  It struck me as odd that it would
truncate a 0 from the value being passed...  I didn't find any spare zeros,
just a bunch of filament and bits of failed prints.

-Aaron

GregValiant, 2022-10-12:
This is one of those things where if it was happening a lot there would be screaming and gnashing of teeth.  I haven't heard any of that.
I think the line goes in as the transition from "Initial Printing Temperature" to  "Printing Temperature".  It seems like it always goes in whether they are different values or the same value.
If your Initial Printing Temperature was 200 and the typo (maybe, possibly, coulda, might've) been in the Print Temperature box then that would have done it.  I don't think there is any manipulation of the setting value in Cura.  I don't know the code but I would assume it just takes the Printing Temperature value and plugs it into that line.
I'll leave this as a bug report as it shouldn't take long to check the code in Cura.

aaron-casper, 2022-10-13:
OMG, I feel so silly now.

[image: image.png]
I totally borked it, and didn't think to check my settings before filing a
report.

Thanks for the second set of eyes!
-Aaron

GregValiant, 2022-10-13:
Not a problem.  At least I didn't have to explain for the 27th time why the Crystal Snake won't slice.
Figuring out what went wrong is the big thing.
If you are confident it was a case of the dreaded "user error" can we close this report?

PS:  As a reward for paying attention here are some extras in case you need them in the future.  
0 0 0 0 0 0

aaron-casper, 2022-10-13:
Yep, the report can be closed.  I'll put those spares in a safe spot for
later.

Thanks again
-Aaron

closed 2022-10-13 by GregValiant