Ultimaker/Cura #13537: Profile List Inconsistencies

ianrossg, 2022-10-12:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ultimaker 2+ Extended

### Reproduction steps

1. Import 3mf file referencing one of the profiles in the two lists.

### Actual results

As can be seen in the screenshot, the Profile List will be different depending on what was imported. The software will otherwise launch by choosing one of the two lists.

### Expected results

I would expect all the created profiles seen in the /quality_changes/ folder to be listed in the Profiles List.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

ianrossg, 2022-10-12:
I had a second dummy printer in Cura, never used, and when switching to that printer the alternate profiles appeared. All the printer settings were identical, so I'm not sure what would have caused this. I deleted the second profile and restarted the software, but unfortunately that did not consolidate the profiles into one list. Importing the 3mf file utilizing one of the alternative list profiles was still able to swap the profile lists.

Is there a way to consolidate the configuration files manually?

MariMakes, 2022-10-25:
Hey @ianrossg 

Thanks for your report 👍 
Sorry, it took us a while to get back to you 😞 

[image]

I believe you are only seeing profiles that are based on profiles that are available. 
That could be related to material or nozzle. 

In your first screenshot, you can only see only profiles based on Normal are visible (nozzle size 0.6 mm), in your second screenshot profiles based on Draft are visible (nozzle size 0.8 mm) 

If you open a file in the quality changes folder you see that this profile is based on 
Definition = Creality Base and Quality Type = Super 
[image]

I don't recommend it but in theory, you can adjust these values to make them visible for other configurations. 

I don't believe this is actually a bug but you are seeing intended behavior. 
Would you agree that this issue can be closed?

ianrossg, 2022-10-25:
Yes, thanks. I too realized this was the case around a day ago and have since been trying to figure out if I've actually been changing my nozzles as much as I probably should be, lol. Thanks, appreciate the time spent toward diagnosing this.

closed 2022-10-25 by ianrossg