Ultimaker/Cura #13540: prevent sleep mode during usb printing

murataktaran, 2022-10-12:
### Is your feature request related to a problem?

I need prevent sleep mode during usb printing

### Describe the solution you'd like

When i printing over usb, windows goes sleep by power management settings. I don't want disable windows sleep settings. Please add a feature about solve this problem. Thanks.

### Describe alternatives you've considered

When i printing over usb, windows goes sleep by power management settings. I don't want disable windows sleep settings. Please add a feature about solve this problem. Thanks.

### Affected users and/or printers

Afected all printers.

GregValiant, 2022-10-13:
UM Cura is open source and anyone can make changes to the program.  In the case of USB printing, no UltiMaker printers use it and so the UltiMaker Company won't put any effort into it and it is up to someone else to make any changes.

I don't know that there is any way to keep a Windows computer from going to sleep without going into the power management settings.  It is the same with disabling Windows Updates which often happen at night and can disrupt a print when the computer shuts down and re-starts.

As far as I can tell, Microsoft does not allow either setting to be set via code.  Someone could write a virus and your computer would go to sleep every 1/2 second.

prostolyubo, 2023-05-31:
The same happens on OSX,
Cura, when printing via USB should prevent sleeping (which is supported by the OS)
[image]

mspoorendonk, 2024-01-11:
I wrote this wrapper around cura based on this code: https://superuser.com/questions/1642810/prevent-pc-from-falling-asleep-when-a-certain-program-is-running It prevents the PC from going to sleep.

'curaNoSleep.vb
' this prevents the system from going to sleep while running cura

' first compile this to exe with the following command (The Fusion 360 print dialog expects an exe)
' C:\Windows\Microsoft.NET\Framework\v4.0.30319\vbc "curaNoSleep.vb" /out:"curaNoSleep.exe" /target:winexe

imports System.Runtime.InteropServices
Public Module MyApplication 
Public Declare UNICODE Function SetThreadExecutionState Lib "Kernel32" (ByVal esFlags as Integer) as Integer
Public Const  ES_AWAYMODE_REQUIRED = &h40
Public Const  ES_CONTINUOUS = &h80000000
Public Const  ES_DISPLAY_REQUIRED = &h2
Public Const  ES_SYSTEM_REQUIRED = &h1
Public Const  ES_USER_PRESENT = &h4

 Public Sub Main ()
  Dim wshshell as Object
  Dim Ret as Integer
  WshShell = CreateObject("WScript.Shell")
  Ret = SetThreadExecutionState(ES_Continuous + ES_System_Required + ES_Awaymode_Required)
  WshShell.Run("""C:\Program Files\UltiMaker Cura 5.6.0\UltiMaker-Cura.exe"" """ & Command() & """", , True)
 End Sub
End Module

The hardcoded cura path is a bit crude. Does anyone have a suggestion for improve ment such that it keeps working when a new version of cura is installed?

alessandro-antonelli, 2024-11-27:
Is there any progress on correcting this embarrassingly elementary shortcoming?

GregValiant, 2024-11-27:
None that I've seen around here.
Ultimaker doesn't use USB printing and they aren't going to pay their people to fix something for my Creality printer.
That's something that I completely understand.

Doing some searching I came across a reference to "PyAutoGUI" that would appear to be the type of thing that is needed.  According to the documentation it can be configured to send a virtual keyboard signal (Ex: "Shift" key press) every once in a while so the computer doesn't go to sleep.  It appears to be some sort of add-on to Python and is not included with a regular Python package.

There is also the option to use Octoprint.  I'm not familiar with it myself.  I use "Converse All-Star mode" to get the gcode to the printer.

fohrloop, 2025-03-18:
Hi, I'm the author of python package called [wakepy](https://github.com/fohrloop/wakepy), and I think that it could be used here for the task of preventing sleep mode. It's cross-platform and it uses the native APIs of the system to keep the system awake. It's otherwise dependency free, but on Linux it depends on [jeepney](https://pypi.org/project/jeepney/) for D-Bus communication. The jeepney package seems to be listed already in [conandata.yml](https://github.com/Ultimaker/Cura/blob/0273ef49f192f59cf8291cb9b631af218dd30f41/conandata.yml#L269) in `pip_requirements_core` (not sure if that's the place for the main requirements of Cura, though).

GregValiant, 2025-03-18:
That looks to be helpful.
Most of the Cura users aren't very computer savvy and would struggle with installing "wakepy" and integrating it into USB Printing.  Although there have been a few reports of problems with Linux and MAC, the vast majority of problems seem to be reported for Windows systems.

The Cura team is currently busy preparing for the release of 5.10 stable, but "wakepy" appears to be something that might be very helpful in getting USB Printing back on track.