Ultimaker/Cura #13547: (5.1.0) Duplicated Materials not Assigned Unique GUID

id-noblis, 2022-10-12:
### Application Version

5.1.0

### Platform

macOS Catalina

### Printer

Ultimaker S5

### Reproduction steps

I created an initial material profile and then duplicated it several times, changing the labels, colors, etc. for all the colors of a particular type and brand of material that I have.

### Actual results

Every material has the same GUID.

```
% cat custom_material* | grep GUID
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>
    <GUID>6196d2de-ad46-47e2-9e21-c0baa5006458</GUID>

### Expected results

Duplicating materials should result in the creation of a new GUID for each material, especially if at least one parameter is changed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

While duplicate materials have the same GUID, they are unable to be synced with Ultimaker printers. Only the initial material with that GUID can be synced. This means that several colors of a material duplicated from an initial material profile must be "Unlinked" for them to be assigned new GUIDs to enable them to be synced. I still have yet to understand the purpose of linked materials in the first place. It is not the case that updating the print settings on one material results in updating the print settings for all linked materials, which is what one would reasonably expect.

MariMakes, 2022-10-13:
Hey @id-noblis,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report. 👍 

Your observations seem to be by design. 
The GUIDs of the materials stay the same until they are unlinked. 
[image]

When you Unlink them they will be treated as individual material but you will lose any potential improvements that we might introduce in new versions of Cura or improved deprime settings in new versions of the Firmware. 

I'm super curious 🤔 
Can I ask why you are creating these new materials with new colors?
Creating 11 materials seems like a lot more than I usually see users do.

MariMakes, 2022-10-13:
Hey @id-noblis,

Ah, if it's just the colors you are concerned with you can also select a color of the Ultimaker Material you are trying to print with. We have 11 colors of PLA. You can also select them in the printer if you don't have the NFC chips if you scroll all the way to the bottom of your selection list if you are on the latest firmware.

[image]
[image]

I usually suggest this workflow over creating new materials if it's a common material. 
What kind of material are you printing?

id-noblis, 2022-10-13:
@MariMakes, it's not an Ultimaker material, it's another brand of PETG. I'm creating new material profiles for each color so I can replicate the workflow as though it were an Ultimaker material, where colors are accurately depicted in Cura and selecting the material in Cura and the material station makes it easy for the printer to load it up. I have tried using the generic and Ultimaker PETG profiles but they have not been successful for this brand. So while I am tuning the print settings for this brand of PETG, having linked materials, where updating the print settings of one material automatically changes the print settings of all linked materials, would be very useful. But as far as I can tell, there is no practical use for linked materials, and the implementation of material GUIDs is misguided.

Reynolds-Beta, 2022-12-11:
I'm in agreement here. I'm using MHBuild PLA which has a number of other colors besides the core Ultimaker PLA (like Silky Copper and Bronze). These have a different temperature range than the Ultimaker PLA or the generics - or at least they benefit from a lower-than generic temperature range.

My problem is even if I unlink them (which does give a unique GUID), the variants do not get synchronized to the Ultimaker S5. In fact, there doesn't seem to be a decisive way TO sync them - "Sync Materials" just spins, never completing, and "Sync via USB" doesn't work either. 

Finally, if I use the local Ulimaker API, pushing a new material has a non-optional "signature file" input which has no documentation.

[MHBuild material profiles.zip](https://github.com/Ultimaker/Cura/files/10202902/MHBuild.material.profiles.zip)

Reynolds-Beta, 2022-12-11:
> When you Unlink them they will be treated as individual material but you will lose any potential improvements that we might introduce in new versions of Cura or improved deprime settings in new versions of the Firmware.

It depends on the implementation, of course. IF there is a change to the generic profiles, then linked profiles should get those updates, yet still override based on the user's preferences in that linked profile. If there is a conflict, then there should be an obvious notification to the user.

Imagine this scenario: Define a new material based on generic PLA, with distinct setting for the particular new material being used (like temperature range). At some later point, generic PLA gets some update (perhaps to retraction settings). This change flows down through any linked profile where it doesn't conflict. Where it DOES conflict, user is presented with obvious UI which describes the conflict, where the conflict came from, and a choice to resolve the conflict.

If user chooses to keep their settings in their custom profile, then the profile takes all other non-conflicting settings from the updated generic PLA, then unlinks - keeping their overridden settings.

If the user chooses to take the settings from the updated generic PLA, then their custom profile remains linked, taking all of the updated settings from the updated generic PLA.