Ultimaker/Cura #13548: Backend exited abnormally with return code 3221225477 when slicing

elaverick, 2022-10-12:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Slicing a model of a greyhound on Super Quality

### Actual results

I get a back end error and it fails to slice

### Expected results

It should slice the model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9767377/cura.log)
[GreyhoundBased.zip](https://github.com/Ultimaker/Cura/files/9767380/GreyhoundBased.zip)

GregValiant, 2022-10-12:
Thanks for the report and the project file.
This is a popular error in the Windows version of Cura 5.1.  The bug was worked on and the changes implemented in 5.2beta.  I would suggest giving that a try.
I'll mark this as a duplicate of #13309 and #12362 and close it as it appears to have been fixed.

closed 2022-10-12 by GregValiant

MariMakes, 2023-02-23:
Hey @elaverick 

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors.
[image]

I was able to slice your model in the beta. Can you download the Beta and retry your slice?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, please let us know

Mulakulu, 2024-01-25:
I am still experiencing the error in 5.6.0, but only when using Tree supports

Paradroid09, 2024-01-28:
Hey @all,

I'm facing the same problem. Also when using treef supports.

Hope it helps to isolate the error.

Mulakulu, 2024-01-28:
I found it to work on the alpha version 5.7 @Paradroid09