Ultimaker/Cura #13554: Cura is closing unexpectedly

GiseyE, 2022-10-14:
### Application Version

5.1.1

### Platform

Windows 10 Business

### Printer

New defined printer or any existing printer

### Reproduction steps

-Uninstall older versions and install 5.1.1
-Load any stl
-Start working on the printing settings
-Cura is crashing

### Actual results

-Cura is closing unexpectedly 

### Expected results

-Cura is working

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9790421/cura.log)

rburema, 2022-11-09:
What do you mean exactly by 'working on the printing settings'?

Is this an operation within Cura itself, or are you editing the print-setting files life?

GiseyE, 2022-11-14:
> Hey @GiseyE<https://github.com/GiseyE>,
>
> Welcome to the Ultimaker Cura Github 🚀
> Sorry, it took us a while to get back to you 😞
>
> I can have someone look at your logs to see what is causing your Cura to crash.
>
> Did you have a chance to upgrade to the latest version of Cura 5.2.1?
> Are you encountering the issue there as well?
>
>
> EXTERNAL EMAIL: Be cautious, do not click links/attachments. Never give out your user ID or password. Verify the recipient's validity
>
> Company Disclaimer: The information contained in this communication is confidential, private, proprietary, or otherwise privileged and is intended only for the use of the addressee. Unauthorized use, disclosure, distribution, or copying is strictly prohibited and may be unlawful. If you have received this communication in error, please notify the sender immediately.

Hello,

Thank you for the follow up. All is working now; I had to uninstall everything and delete some files manually, that were not being uninstall/deleted automatically. Also update some visualization stuff in my computer.
Then install again. Now is working

Thank you,
IGEW

rburema, 2022-11-15:
Thanks for getting back to us!

Could you tell us which files where the problem?

rburema, 2022-11-17:
Since all seems to be working now on your end, I'm closing this issue. If unexpected files (so not the user data for example, but things in program files) stay after deinstalling again, please feel free to open a new issue.

closed 2022-11-17 by rburema