Ultimaker/Cura #13555: Support Blocker Option Related to Support Base

RealDanCarroll, 2022-10-14:
### Is your feature request related to a problem?

I incorrectly assumed that the support blocker could be used to prevent the floor of Tree supports from being rooted in certain locations. I was wrong and there appears to be no good way of indicating where the root of a tree support is placed beyond being able to indicate "Touching Buildplate" or "Everywhere".

I find Tree Supports useful. Sometimes the root must fall on the model instead of touching buildplate, though i prefer the root to touch the buildplate. When i am forced to use "Everywhere", I would love to have some control over where Tree supports do and dont spawn.

### Describe the solution you'd like

Allow me to use something similar to a support blocker... but we call it a "Tree Support Root Location" I could then place an indicator on the buildplate or on the model where I want the tree supports to generate from. If this feature is developed, you should be able to paint on the area or create multiple root spawn areas.

### Describe alternatives you've considered

A faster/easier to develop solution that wouldnt be as robust or useful but would still give me the power I desire.... I would like to be able to place a support blocker on the model and have it function to prevent the root of a tree support from being generated in that particular part of the model when "Everywhere" is selected.

### Affected users and/or printers

Anyone that uses your wonderful Tree Supports could benifit from this feature.

DerGenaue, 2022-10-20:
Check out this [alternative Tree Support implementation](https://github.com/ThomasRahm/CuraEngine);
It does most of the things you described

RealDanCarroll, 2022-10-20:
If it works, Cura shoud integrate this into the release. Im definetly going to give this a look.

RealDanCarroll, 2022-10-20:
Closed by accident.... a 3rd party suggestion shouldnt close the issue.

DerGenaue, 2023-07-04:
The new algorithm has been integrated into the new Cura 5.4 👍