Ultimaker/Cura #13556: Disabling auto-scaling on import of 3MF still auto-scaled object to 10000%

ansonl, 2022-10-15:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

UMO+ DXU Dual

### Reproduction steps

1. Turn off Scale large models and Scale extremely small models in Preferences
2. Import a 3MF file that has 2 objects. 1 of the object has some intentionally small features and the X/Y/Z coordinate of the objects relative to each other are important. 
3. Cura auto-scales the imported 3MF file and both objects in 3MF file are scaled to 10000%.  The X/Y/Z coordinates of the objects are also lost upon manually changing scale back to 100%. 

This problem does not happen with OBJ files, but separate objects in a single OBJ file are not independent in Cura and are imported as a combined object which defeats to purpose of 2 objects in a single file with set X/Y/Z coordinates. I read at https://community.ultimaker.com/topic/29700-do-not-auto-orient-when-importing/ that Cura respects 3MF file coordinates which is important for my use case. 

### Actual results

Cura auto-scales the imported 3MF file and both objects in 3MF file are scaled to 10000%
autoscaled
[image]
after setting scale to 100% manually, object locations are lost
[image]

### Expected results

Cura imports the 2 objects in the 3MF file with auto-scaling and retains the original X/Y/Z coordinates of the objects relative to each other. 

correct import of 3MF file with 2 objects and correct coordinates in Microsoft 3D Builder
[image]

import of OBJ file in Cura but 2 separate objects are imported as a single unmergable object
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The imported object CT-combined.3mf contains an object with intentionally thin wall.
It's not super important for this bug, but purpose of the thin object is have a 1 wall thick Gcode generated at the X/Y/Z coordinates relative to the other object coordinates within the 3MF file. The thicker object would be deleted and the thin object be sliced and Gcode postprocessed. 

[cura.log](https://github.com/Ultimaker/Cura/files/9791106/cura.log) The imported file in the log is CT-combined.3mf attached below
[cura-project.zip](https://github.com/Ultimaker/Cura/files/9791107/cura-project.zip)
[CT-combined.zip](https://github.com/Ultimaker/Cura/files/9791104/CT-combined.zip) The original CT-combined.3mf

GregValiant, 2022-10-15:
Thanks for the report.
If there are multiple objects in the file you are importing and they are grouped in Cura you can use CTRL+Left CLick to pick out one model from a group.
You could also select the group, right click on it, and select "Ungroup" from the menu.
When I opened the project file the parts were separate pieces but were both "selected" and so moved as a group.  I clicked on the build plate and then re-selected one of the models and it moved independently.

Regarding the scaling issue - I don't know.  Your project file imports the same way for me scaling the file to 100,000%.  I have the Mesh Tools installed and the import scaling is set to millimeters.  Like you, I have auto-scaling turned off in the preferences.

This may be a Cura bug and I'll leave it as a bug report.  The fact that there are two possible errors here in a single model file may point to the export utility and the way the OBJ/3MF are being created.  Someone from the Cura team will take a look.

ansonl, 2022-10-15:
@GregValiant Yes, I am able to group/ungroup the 2 objects in the 3MF file as well. I had meant that 2 separate objects in an exported OBJ (from blender) can't be ungrouped but since @Ghostkeeper maintains a 3MF exporter plug-in for blender and cura is supposedly respects 3MF object coordinates, using 3MF instead of OBJ is a workaround for the unrelated grouping issue.

LeeLorenzSr, 2023-04-18:
I had a model auto-scale in a 3mf file. This problem is driving me crazy. If I have scaling turned off, it should NEVER scale, and it certainly shouldn't scale one component of the 3mf and leave the rest the same scale. Auto scaling should have the ability to be completely turned off. I've had this terrible "feature" insist itself on stl and 3mf file imports. It's been an issue for at least 8 years, according to google.

It also moves components on import, which is TERRIBLE for multi-extruder slices, since I need those components precisely positioned in relation to one another. I'm not sure what is going on with this, but it feels like the entire logic Cura imposes on positioning and scaling objects being imported is psychotic, with objects positioned off the bed (0,0 and settled on floor plane in the design software).

nallath, 2023-04-19:
For multi extruder slices, you are supposed to merge the two objects together. This forces the models to use the same origin. You can do this by selecting them and selecting the option via the context menu.

3MF files positioning will be respected. What can happen is that the "Drop to buildplate" option is enabled, so you might see that the models are loaded and then dropped down. You can also disable this if you want to. All autoscaling can be turned off. 

The autoscaling in this case isn't what causing the issue. The project file that were originally added have the scaling set to that number in the 3mf themselves. Cura is respecting what is stored in those 3mf files. I've also loaded these models in other 3mf viewers, and they indeed seem to be ~65000 mm wide. 

I understand that you're frustrated, but I would like to ask you to consider changing the tone of your messages. We're trying to help here, there is no need to talk like this.

LeeLorenzSr, 2023-04-19:
> For multi extruder slices, you are supposed to merge the two objects together. This forces the models to use the same origin. You can do this by selecting them and selecting the option via the [...]

Sorry for the tone. I did try the "merge" option on the multi-extrusion project, but it did not reset the components to their respected origins. When this does not happen, you can understand how frustrating it is for a user who sees the instructions on doing this all in many videos and guides. I am using 3D Builder to do quick boolean operations to cut out, then replace a section of the model with a copy of the part used to trim... If I had more time, I'd be happy to contribute; I am a professional developer with 40 years of experience.

Just to run a sanity check, I just tried to perform the "merge" operation on a multi-extruder project, and it rotated the main object 90deg on two axis, and placed the smaller object at 0.0 origin, above the build plate. Definitely not correct, I will take some time to document this, and post an issue if I can.

As for this issue, the auto scaling, I have both scaling options turned off. Why is this not sufficient to disable the transformation from happening on import? Is there another setting I need to disable/enable?

nallath, 2023-04-19:
The only scaling that i've seen happen is that it uses the scaling as stored by 3D builder, but that is based on the files originally posted above.

So what is basicly happening is that your 3D model has a certain size. Let's call that 100%. In 3MF you can then add that model as an object, but an object can also have a scale (which is the scale WRT to the model). So if you have a model that is 10x10x10mm, and you add it to the buildplate as an object that is scaled by factor 10, Cura will load this as an object that is scaled 1000%, as that is what the file tells it to do. I also vaguely remember that we might do some scaling depending on the unit (as 3MF actually defines the unit, but Cura needs mm), but I'm not 100% sure if that would affect the scaling in a way that the user can see. 

Once you start scaling / rotating things in Cura, the merge gets hairy. This in itself is a tricky problem, so I'm not sure if you're running into an actual bug or just something that is inherently unsolable (as it's behavior is amigious). 

If you have "drop to buildplate" and "scale large/small" objects turned off, the 3MF should load the exact position of the models. If that doesn't happen; It's a bug and we should fix that.

LeeLorenzSr, 2023-04-19:
I'll try and dedicate some time to documenting the issue better, with screenshots. 3D Builder is pretty basic, but for some of us, it's quick and easy, and does a pretty good job of repairing models. I imagine it's a workflow for many Cura users.
In my previous example, it's a model I scaled up 147% in 3D Builder, positioned at 0,0 origin, and rotated on one axis, so that if it's not repositioned by Cura, it would sit across a build plate. It's a model of the side of the Tardis (My wife is a whoovian, and I'm modifying it to be a 300mm+ tall box). The window panes in the model are open, I'm placing a flattened (white PLA) cube behind the frame to fill that space. I size and position the other object, make a copy, trim behind the frame, and then place the copy back in that spot. I note the position, relative to 0,0 for positioning in Cura. Dropping the 3mf into Cura is where it gets interesting. As a developer, I know recreating a problem consistently is key to tracking the problem in code, so I will try and create a pure set of files that can be run - same goes for this rescaling issue. I do feel like they are related.

ansonl, 2023-06-13:
@nallath I tried importing a 3MF file that is 2 STLs exported from Blender and still had the same scaling issue. I included screenshots of my 3MF export settings in Blender, Cura settings, and the imported scale of the objects in Cura. I am using the 3MF format add on for Blender by GhostKeeper https://github.com/Ghostkeeper/Blender3mfFormat

A test file where this scaling issue occurs despite scaling and drop settings disabled is attached:
[CT.zip](https://github.com/Ultimaker/Cura/files/11740208/CT.zip)

[image]
[image]

[image]

ansonl, 2023-07-17:
@nallath Bumping since the issue still exists in 5.4.0

dmandn, 2023-09-03:
I confirm this is happening in Cura 5.4.0.
Issue presents with 3mf files from Fusion 360, tested on multiple printers. My files keep scaling down approx 10%