Ultimaker/Cura #13558: Are the three new Initial Layer Flow settings additive (really, multiplicative) or Overrides?

Ergonomicmike, 2022-10-15:
### Application Version

5.2.0 Beta

### Platform

Win10 64-bit

### Printer

Ender 3 V2

### Reproduction steps

Not a Bug Report, per se. More of an Ergonomic Issue understanding the three new settings that are in 5.2.0.

[image]

My question is, are these new settings affected by the Initial Layer Flow setting (multiply each another) or are the settings independent Overrides?

For example, after loading the stock "Standard Quality" defaults for a printer, if I set the Initial Layer Flow to 123%, then, automagically, all three of the new sub-Layer Flows update to show 123%.

[image]

From this behavior, I'm inclined to conclude that the Initial Layer Flow is simply a "Quick Set" placeholder, which does not affect the three new sub-Flows below it.

BUT, since the popup Help says that Initial Layer Bottom Flow is affected by the Initial Layer Flow, I have to wonder if that means that all the sub-Flows will be 1.51% (1.23% x 1.23%)?

[image]

### Actual results

I don't know. I suppose I should have tested more before filing this Issue by setting one of the sub-Flows to 0% to see what happens during a print.

### Expected results

I think I expect all three Flows to be independent of the Initial Layer Flow.

If that's the intent, then I suggest that the UI needs to give Users visual feedback about this.

For example, if a User uses the Initial Layer Flow to set all three sub-flows to the same value, then the screen shot above is fine. (Where all the sub-flows populate to 123% after the Initial Layer Flow is set to 123%.) This feedback would suggest to the User that the new Inner, Outer and Bottom Flows are each going to be 123%.

And then, if any of the three new sub-Flows is changed, then I suggest that the UI gray out (or black out) the top most "Initial Layer Flow" to show that it is no longer controlling.

(Or maybe easier, just eliminate the Initial Layer Flow setting altogether and Users will only have to deal with the three new sub-Flows.)

However, if it's the Devs' intent that these numbers multiply each other, then there might need to be a "Total" field that pops up, to show a User that if they have set the Initial Flow to be 123% and if the User leaves the Bottom flow at 123%, then the total Bottom flow will be 1.51%. And so forth for different percentages.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Whichever behavior is intended, it would help if the Help made it clearer.

GregValiant, 2022-10-15:
Thanks for the question.
From the commit by @Vandresc that added (or made visible) those settings:
_"Added the new three **children** settings of Initial Layer Flow to Expert mode to match the parent."_

So it would appear that you can set them all together when they are linked to the "parent" control "Initial Layer Flow", or you can set them separately which un-links them from the parent.
They don't add or multiply but rather stand on their own.

_"Whichever behavior is intended, it would help if the Help made it clearer."_
That is true of many of the help balloons.  This happens when English isn't the native language of the software engineer.

Ergonomicmike, 2022-10-15:
Okay, thanks. Good to know.

closed 2022-11-01 by Ergonomicmike