Ultimaker/Cura #13559: Brim order completely out of wack

CyberFranck, 2022-10-15:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Custom (Ender 5 pro)

### Reproduction steps

Activate Brim

### Actual results

Brim does 2-3 first outer lines then jumps 2 and keep going for 5-6 then jumps 1 then finish up to where the part starts then come back fill the missing spot. I have attached a video showing the simulation. Everything else print properly but this jump is really not helping.
[2022-10-15 09-08-51.zip](https://github.com/Ultimaker/Cura/files/9792134/2022-10-15.09-08-51.zip)

### Expected results

I expect the brim to work as it used to in previous versions

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no additional info

GregValiant, 2022-10-15:
Thanks for the report.
This is a duplicate report of the behavior. I believe this was fixed for 5.2beta.  You could give that a try.
I'll close this as a duplicate of #11936 and #13245.

closed 2022-10-15 by GregValiant

CyberFranck, 2022-10-16:
Well brim is fix indeed but the part layer order is now broken. Time to downgrade to 4.x I believe
[2022-10-16 11-23-06.zip](https://github.com/Ultimaker/Cura/files/9795022/2022-10-16.11-23-06.zip)

CyberFranck, 2022-10-16:
Ok 4.13 is the solution. It works perfectly.

GregValiant, 2022-10-17:
That part layer order looks OK to me.  Inner Walls, Outer Wall, Inner Wall, Outer Wall, Middle Wall.  In 4.13 the middle wall would be skin.
I liked 4.13 myself and I'd gotten very good with it.  5.x is the new deal though so I strive to figure it out and get it printing the way I want and I do believe I've got it (except for the silly motion of the Monotonic Layers in 5.2beta).  It was a huge leap for the Cura developers moving Arachne in and moving to QT6 dialogs all at once.  It's getting better and the dialogs are working better.  The additional settings mean there is a learning curve for everyone and there has been a lot of push back about that.