Ultimaker/Cura #13560: Slicing error when using "Spiralize"

GregValiant, 2022-10-15:
### Application Version

5.2beta

### Platform

Windows 10 Home 64bit

### Printer

Ender 3 Pro

### Reproduction steps

This report started with a thread on the UM Forum [HERE](https://community.ultimaker.com/topic/41786-vase-mode-seam-and-artefacts/#comment-313726)

Loaded a suitable model into Cura.
Set up to spiralize with .8 nozzle and .2 layer height at .8 line width.
Natural PLA.
There are localized bulges in the print.  They were not apparent in a print when using opaque filament.

### Actual results

The print has a couple of areas with localized bulges.
[image]

Here you can see that they are in the gcode.  The max offset is 0.1478mm.
[image]

This is a screenshot of a second gcode file as I was setting the Project File to upload here.  The "waves" in the lines show up in the print.
[image]

### Expected results

A flawless print with a finish like 12 coats of hand rubbed lacquer.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[GV_Spiralize_Report.zip](https://github.com/Ultimaker/Cura/files/9792281/GV_Spiralize_Report.zip)
I did not include a log file.

smartavionics, 2022-10-15:
Changing the Maximum Deviation to 0.1 (was 0.025) made a big difference to the output, you may find doing that yields better results...

Using 0.025 we get lots of segments on some layers (including the region that makes you unhappy)

[image]

[image]

Using 0.1 reduces the segment count and may print nicer (not tested :)

[image]

[image]

GregValiant, 2022-10-16:
Thanks @smartavionics.  That worked and the test print turned out excellent.

closed 2022-10-16 by GregValiant