Ultimaker/Cura #13565: Monotonic enabled, excessive travel

frankygoop, 2022-10-15:
### Application Version

5.2.0

### Platform

Windows 11

### Printer

Tevo Tarantula

### Reproduction steps

Enable monotonic, go to layer 33 

### Actual results

observe the excessive travel over the wall skin. It made a mess on the outer wall. I recorded a video but could not uploaded due to the size.

### Expected results

to behave like with previous versions.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[TT_Celholder.zip](https://github.com/Ultimaker/Cura/files/9793297/TT_Celholder.zip)

GregValiant, 2022-10-15:
Thanks for the report.
You're being nice about it.  It's a ridiculous amount of travel.
I believe that the 5.2 stable release will have Monotonic back the way it was.  Besides the travel there is a lot of stringing caused by this motion.
I'll remove the bug label as this is Expected Behavior in 5.2beta.  Hopefully what I heard about reverting back to the previous motion will turn out to be true.  @MariMakes may have an update on that.

MariMakes, 2022-10-17:
Hey @frankygoop,

Thanks for the report 👍 
@GregValiant is right, it's a known error in monotonic on our side that has been resolved in https://github.com/Ultimaker/Cura/issues/13456

I checked our latest internal version and you can see the fix, and the immensive time save, on the right. 
[image]

Your issue should be fixed in 5.2 Stable once it is released. 
Thanks again for the report 😄 
I'll be closing this issue.

closed 2022-10-17 by MariMakes