Ultimaker/Cura #13566: Changing material profile does not transfer new temperature setting to main printing profile

tkocou, 2022-10-16:
### Application Version

5.1.1

### Platform

Linux Mint

### Printer

Anycubic Vyper

### Reproduction steps

I physically loaded the new filament and went to Cura looking for that material. It was not listed. I opened the Materials menu and clicked on "Create new". I made all the necessary changes for the loaded filament. Then I went to the triple line menu and clicked on "activate". The display showed the new material, however, when I checked the printing profile, the Material area showed the old settings from the previous filament.

### Actual results

I changed the printing profile Material area to match the new filament

### Expected results

The current printing profile should have picked up the new settings for the selected materiral

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

cura.log
[cura.log](https://github.com/Ultimaker/Cura/files/9794328/cura.log)

GregValiant, 2022-10-16:
Thanks for the report.
If you are using one of the "canned" quality profiles (Fine, Standard, Dynamic, etc.) then the Material will override the Profile.
If you are using a Custom quality profile then the Custom setting will override the Material settings.
This is one of those things where no matter how it is set up, someone would complain about it.  That might even be how it ended up the way it is.
This is a duplicate of many other reports both for doing it the way you would like, and doing it as it is (with "Custom Quality Profiles" being at the top of the override hierarchy).
The developers might review the way it is.

tkocou, 2022-10-16:
Why not add an option to the preferences which will allow one to tell Cura "Always apply selected material settings" regardless of whether they are using a canned profile or a custom profile. I always create a new profile whenever I make changes to the printing profile. That way, I always have the canned profile to start with in case I messed up the settings in the custom profile.

CoryCharlton, 2023-01-20:
> Why not add an option to the preferences which will allow one to tell Cura "Always apply selected material settings" regardless of whether they are using a canned profile or a custom profile. I always create a new profile whenever I make changes to the quality profile. That way, I always have the canned profile to start with in case I messed up the settings in the custom profile.

I was just coming to add the same feature request. @GregValiant is correct that no matter which default is done someone will not be happy (ex: you and I are not happy that the profile overrides the material). 

So having a general option that allows the user to select the setting priority for either the material or the profile is the best solution.

@GregValiant would you recommend that a feature request be opened for the new setting priority setting being suggested here or does this issue work?

GregValiant, 2023-01-20:
A feature request would be the way to go.  This is one of those things that I call "Give it some love".
Since the release of 5.x I think the Cura team has been running around putting out fires.  When they get a chance to take a breath they can start going over the GUI and controls and make them more user friendly.  There will never be an "Easy" button in Cura, but some settings are tough to find, or duplicated in about 4 places and some update and some don't.  A little plain old housework would seem to be in order and part of that would be feature requests like this.  I find the entire "profile" thing baffling.  I have three profiles and some days that's two too many.

Kotoubiyca, 2023-09-19:
> An alternative solution: Install the "Material Settings" plugin, that adds the option to "**use the value from the material**" to the right-mouse menu in the print settings list.
> 
> Then the parameter will link the settings present in the Material definition with your current settings by using the extruderValueFromContainer instruction.

Can you post the link to this plugin? [https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/](MaterialSettingsPlugin) say a 'This plugin does not change the fact that if a setting value is specified in the "sidebar" settings or in a quality profile, this always overrides the value set for the material.'

GregValiant, 2023-09-19:
That's part of the "Override Hierarchy" I mentioned.  Your main "Custom Setting Profile" is the top of the pile and any value you add there will be used.  It can be "Linked" to the Material Settings, or it can be a custom number for "this print".  You can revert by re-selecting the Custom Profile and "Discarding Changes".

If your material has a Print Temperature of 205° but for _this little print_ you want it at 200° you would type in 200.  That would be used in the slice and if you create a "Custom Profile" with 200 as the temperature, that's what it will be.  It will no longer be linked to the Material Print Temperature in the Material Settings.
Next to the setting box will be an "fx" and/or a "circle arrow" icon.  When you click them - the Print Temperature setting will revert back to "Material Print Temperature".  It will be re-linked to the Material Setting.  If that is the case then if you were to change the Material Setting, that box in the Custom Settings would change to your new Material Print Temperature.

It's confusing even writing this.  It's why I only have the three Custom Profiles and use them for everything.  It's a bit tedious going through all the settings every time to make sure I have them dialed in for "this particular print", but it's nowhere near as annoying as getting 15 layers into a print and noticing that I missed a trick and have to abort.

So a problem that some people see here is that the Material Setting becomes un-linked.  But, there is a system in place to re-link it.
I suppose a setting to insure that the Material Setting table is always enforced would help with this.  I just don't see it being any less confusing.
"Fast PLA", "Slow PLA", "PETG" is all I require as Custom Profiles.  I have a second Ender 3 Pro set up for TPU because it requires a change in the StartUp Gcode purge line speeds.  So TPU is totally separated because my printer hates it.  (It makes damn fine fuel proof gaskets though.)

5axes, 2023-09-19:
> Can you post the link to this plugin? [MaterialSettingsPlugin](https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/) say a 'This plugin does not change the fact that if a setting value is [...]

Well @fieldOfView  should update his page and add : 'This plugin does not change the fact that if a setting value is specified in the "sidebar" settings or in a quality profile, this always overrides the value set for the material, **unless you use the "Use value from material" function. In this case, the values will be taken from the settings defined in the material.**
[image]

5axes, 2023-09-19:
Note: The "Use value from material" function allows you to replace the value defined by the user with the Cura "
ExtruderValueFromContainer" function. If you use this option, you should see the function code appear when you save your profile or replace the value. In this case, be careful not to cancel this function using "Discard Changes".
[image]

5axes, 2023-09-19:
Note of Note : I've used this principle in one of my plugins (which isn't in the marketplace), with specific management of material parameters. 
[image]

The **Material Settings** function group automatically links all existing parameters in the material definition. It also lets you see which parameters are linked to material settings. 

This plugin is risky to use, as Cura offers no Undo mechanism. It should be reserved for very experienced users.

Kotoubiyca, 2023-09-20:
> Note: The "Use value from material" function allows you to replace the value defined by the user with the Cura **ExtruderValueFromContainer** function. If you use this option, you should see the [...]

Thx so much!