Ultimaker/Cura #13567: Randomly get extremely long print time estimates in Vase Mode.

Ohmarinus, 2022-10-16:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Custom profile (Replicator 2x)

### Reproduction steps

It doesn't just happen with one setting, but for example when using my custom profile for vase mode, the print time estimate first was at 32 minutes, but then when I leave `bottom thickness` at 0.6mm and set `bottom layers` to 2, the print time goes up to 494 days.

[image]

This issue goes away when setting `bottom thickness` to 0.

Then I decided to change my print speeds from 100mm/s to 80mm/s and this happens:
[image]

When setting the print speeds to 81, it now goes to show 39 minutes.
[image]

I cannot find out what's causing this issue and I'm working on a design that I cannot make public yet so I can't share the STL file. But I can tell that all of my settings are sane and that I have not had issues with this file and settings in PrusaSlicer nor SuperSlicer.

### Actual results

See reproduction steps.

### Expected results

See reproduction steps.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have just switched from PS to Cura and this is my first profile so I have no way of comparing this result to any other profiles or STL files. There doesn't seem to be a clear reason why this is happening. **I have successfully printed my STL file with the settings that estimate a 39 minute print first try so I do not believe my STL file not my settings are not sane.**

It will be hard figuring this one out, but I believe that this is a serious bug.

GregValiant, 2022-10-16:
Thanks for the report.
I can see where a single model that needs **256 kilometers** of filament and 2.8 years to print might be a problem.  Unfortunately, I can't duplicate this behavior.  Without the model (or a similar model that shows the same behavior) there is nothing for us to investigate.  I'll leave this open but you need to post a Zip folder that contains a project file ("File | Save Project") that shows the behavior you are describing.
Just for comparison:  256826m of filament is about 46 times the build volume of my Ender.

Ohmarinus, 2022-10-16:
> Thanks for the report. I can see where a single model that needs **256 kilometers** of filament and 2.8 years to print might be a problem. Unfortunately, I can't duplicate this behavior. Without the [...]

Yes, I completely understand. I shall try to reproduce this issue with a different model, I'll report back about this in the evening (EU timezone).

Ohmarinus, 2022-10-20:
I've been trying to replicate this issue but it only happens on complex models. However, I did discover that when selecting `spiralize outer contour`, it doesn't set `Surface Mode` to `Surface` but instead keeps it set to `Normal`. This is causing issues in my slicing, adding in unwanted details. Setting it to `Surface` fixed any problem I had with slicing, despite my models being watertight and without issues.

I now start to wonder if this was one of the reasons behind the print time estimates being off.

I did manage to get an even more extreme estimate:
[image]

This happened when I left it on `Normal`. Setting the Surface Mode to `Surface` fixes all issues, but I haven't found out what other factors influence this behavior.

GregValiant, 2022-10-20:
I cannot duplicate this behavior using that "Buggy" profile.  We need a project file in order to investigate.

Ohmarinus, 2022-10-20:
> 

[time_calc_issue_cura5.2.1.zip](https://github.com/Ultimaker/Cura/files/9831489/time_calc_issue_cura5.2.1.zip)

Here is a project with the 3mf file including all profiles and the STL and a log for Cura 5.2.1

This is the calculation I get when selecting 'Normal' instead of 'Surface' for `Surface Mode`:
[image]

GregValiant, 2022-10-21:
That's an odd model and is actually made up of 4 distinct pieces with 0.1mm gaps between the 4 main bodies, but are connected by surfaces at each rib.  That allows Spiralize to work everywhere except for the very top layer.  It's a nice trick and I'll need to remember that.
[image]

Back to the bug...I'm still not able to duplicate your problem.  For reference I'm also on Windows 10.

This is in Surface mode.  51 minutes.
[image]

This is in "Normal" mode with single wall, no top/bottom, no infill and is 53 minutes.
[image]

I keep playing with settings but I can't reproduce the problem.  Even in Normal mode with 100% infill it's just a 25hr print.  That's considerably shorter than the 2 years 9 months that you are getting sometimes.

With the Line Width dropped to .2 you can see the gaps between the parts.
[image]

Those gaps allow this model to be spiralized EXCEPT where the ribs actually span across which only happens on the very top layer.  That's why there is that string on the top layer and nowhere else (when in Spiralize Mode).
In Surface Mode the motion is nearly identical.
Because each rib has an "out and back" extrusion move there will be a flaw in the outer skin at each rib intersection.  That will happen both in Spriralize and in Surface Mode and is there because of the way the model is designed.

GregValiant, 2022-10-21:
Alright, I tried again and now I'm able to reproduce this.
With:
Union Over-Lapping Volumes = OFF
Remove all Holes = OFF
Surface Mode = NORMAL
Spiralize = ON
I'm getting ridiculous numbers.
The problem seems to occur with the order in which Spiralize and Remove all Holes are enabled/disabled.
I'm attaching a project file and a gcode file.  When I open it and immediately slice, the problem occurs...or not.  If I disable "Remove All Holes" the problem goes away.  Re-enabling "Remove All Holes" does not return the problem.
Turning Spiralize on and off with Remove All Holes off can intermittently cause the problem.
[GV_time_calc_issue.zip](https://github.com/Ultimaker/Cura/files/9838569/GV_time_calc_issue.zip)

This preview is of a 1031 day print with the Travel Moves turned on.  The print takes a long holiday as the nozzle drives to Tierra Del Fuego (extruding all the way) before returning in a travel move. That happens on layer:0.
[image]

;TIME:69861104
;Filament used: 200845m
;Layer height: 0.3
;MINX:43.883
;MINY:11.466
;MINZ:0.3
**;MAXX:1.74585e+09
;MAXY:1.74585e+09**

You just don't see this very often:
G1 F600 Z1.1
G0 F9000 X2088993.84 Y2088993.976 Z1.1
G1 F600 Z0.6  -----------------------------------**This is also wrong.**  The error is on Layer 0 so this should be Z0.3
G1 F2100 E213.15066
G1 F3000 X161.395 Y110.305 E200833573.34534

Ohmarinus, 2022-10-21:
> Alright, I tried again and now I'm able to reproduce this. With: Union Over-Lapping Volumes = OFF Remove all Holes = OFF Surface Mode = NORMAL Spiralize = ON I'm getting ridiculous numbers. The problem seems to occur with the order in which Spiralize and Remove all Holes are enabled/disabled. I'm attaching a project file and a gcode file. When I open the project and immediately slice, the problem occurs...or not. If I disable "Remove All Holes" the problem goes away. Re-enabling "Remove All Holes" does not return the problem. Turning Spiralize on and off with Remove All Holes off can intermittently cause the problem. [GV_time_calc_issue.zip](https://github.com/Ultimaker/Cura/files/9838569/GV_time_calc_issue.zip)
> 
> This preview is of a 1031 day print with the Travel Moves turned on. The print takes a long holiday as the nozzle drives to Tierra Del Fuego (extruding all the way) before returning in a travel move. That happens on layer:0. [image]
> 
> ;TIME:69861104 ;Filament used: 200845m ;Layer height: 0.3 ;MINX:43.883 ;MINY:11.466 ;MINZ:0.3 **;MAXX:1.74585e+09 ;MAXY:1.74585e+09**
> 
> You just don't see this very often: G1 F600 Z1.1 G0 F9000 X2088993.84 Y2088993.976 Z1.1 G1 F600 Z0.6 -----------------------------------Double layer height due to Spiralize G1 F2100 E213.15066 G1 F3000 X161.395 Y110.305 E200833573.34534

Did you actually measure the distance from your location to Tierra Del Fuego? That's dedication. I like it!

And thanks for checking this report, I know it kind of started off on the wrong foot with me not supplying enough material to check it with. Now let's see if this is only my model that's causing this issue, or if there's something going on with the slicer.

Oh, and I had this issue again with another model that is created in a very different way, sadly this model is copyrighted and I cannot share it. But if I encounter it again, I might be able to share the model and data if more data is required to get to the bottom of this.

GregValiant, 2022-10-22:
_"Did you actually measure the distance from your location to Tierra Del Fuego?"_ 

I'm still on the road south.  I wanted to check the time so I'm holding my speed down to 180,000mm/hour.  At that speed I'm starting to notice that the rest areas are too far apart.