Ultimaker/Cura #13568: Create a standalone Cura executable (Windows)

yess1987, 2022-10-16:
### Application Version

5.2.0

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
git clone https://github.com/Ultimaker/Cura.git 
cd Cura
conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
.\venv\Scripts\Activate.ps1
pyinstaller .\venv\conan\Ultimaker-Cura.spec

### Actual results

Unable to find "C:\dev\Cura\venv\conan_install_info.json" when adding binary and data files.

### Expected results

I can create a stand-alone frozen Ultimaker-Cura executable and all the dependencies and resources

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

You can create a stand-alone frozen Ultimaker-Cura executable and all the dependencies and resources. Make sure the virtual environment is still active.

pyinstaller venv/conan/Ultimaker-Cura.spec
The build and collected files can then be used in the package manager of your choice, NSIS, AppImage, Deb or you can run things as stated below:

cd dist/Ultimaker-Cura
./Ultimaker-Cura

jellespijker, 2022-10-16:
See the GH Workflow for inspiration:

https://github.com/Ultimaker/Cura/blob/4a30bbaf2b396ccb7eb4aa3920f483ad63e1084b/.github/workflows/cura-installer.yml#L169

Add `--json "cura_inst/conan_install_info.json"` to the install command

yess1987, 2022-10-18:
pip install conan --upgrade
conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force
conan remote remove cura
git clone https://github.com/Ultimaker/Cura.git
cd Cura
conan install . --build=missing --update -if cura_inst -g VirtualPythonEnv -o cura:devtools=True --json "cura_inst/conan_install_info.json"
.\venv\Scripts\Activate.ps1
pyinstaller .\venv\conan\Ultimaker-Cura.spec

this is not work?

DerGenaue, 2022-10-23:
We managed to make it work roughly following the GH workflow:
(note that we load the `Activate.ps1` and the `Ultimaker-Cura.spec` from the `cura_inst` folder)

    conan create . local/testing
    conan install cura/5.2.1@local/testing --build=missing --update -if cura_inst -g VirtualPythonEnv -o cura:devtools=True --json "cura_inst/conan_install_info.json"
    .\cura_inst\Scripts\Activate.ps1
    pyinstaller .\cura_inst\Ultimaker-Cura.spec

Adapting to your approach, does this work for you?:

    conan install . --build=missing --update -if cura_inst -g VirtualPythonEnv -o cura:devtools=True --json "cura_inst/conan_install_info.json"
    .\cura_inst\Scripts\Activate.ps1
    pyinstaller .\cura_inst\Ultimaker-Cura.spec

If the last step fails with a unicode error, you need to go into `\venv\conan\Ultimaker-Cura.spec` and change the `\` in the filepaths to `\\` or `/`

yess1987, 2022-10-24:
after do this:
    conan install . --build=missing --update -if cura_inst -g VirtualPythonEnv -o cura:devtools=True --json "cura_inst/conan_install_info.json"
The following information is not generated.
    .\cura_inst\Scripts\Activate.ps1
    .\cura_inst\Ultimaker-Cura.spec

DerGenaue, 2022-10-24:
then you have to do the route with the local package as described in the first box:
`... create ... local/testing`
`... install ...@local/testing`

jellespijker, 2022-10-24:
@yess1987 

what happens if you remove the whole user data folder, profiles, data etc. from Conan. It looks like our custom generator  
aren't installed,  or that you have old ones.

first remove the following folders:
`C:\Users\Username\.conan` and `C:\.conan\`

Then run everything from the Powershell as a normal user:

```powershell
pip install conan --force --upgrade
pip install sip
conan config install https://github.com/ultimaker/conan-config.git`
conan profile new default --detect
```

now create a temporary folder somewhere cd into that and run
```powershell
conan install cura/5.2.1@_/_ --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_info.json
```
then activate the environment and run cura as a test
```powershell
./cura_inst/Scripts/Activate.ps1
python ./cura_inst/cura_app.py
```

if cura 5.2.1 starts up you can close it and run pyinstaller 
```powershell 
pyinstaller ./cura_inst/Ultimaker-Cura.spec
```
This should create the dist folder with a standalone Cura which you then need to package using NSIS

yess1987, 2022-10-25:
Will the following command still work in cura 5.2.0beta+?
   python ..\cura_inst\packaging\NSIS\nsis-configurator.py ".\Ultimaker-Cura" "..\cura_inst\packaging\NSIS\Ultimaker-Cura.nsi.jinja" "Ultimaker Cura" "Ultimaker-Cura.exe" "$Env:CURA_VERSION_MAJOR" "$Env:CURA_VERSION_MINOR" "$Env:CURA_VERSION_PATCH" "$Env:CURA_VERSION_BUILD" "Ultimaker B.V." "https://ultimaker.com" "..\cura_inst\packaging\cura_license.txt" "LZMA" "..\cura_inst\packaging\NSIS\cura_banner_nsis.bmp" "..\cura_inst\packaging\icons\Cura.ico" "Ultimaker-Cura-$Env:CURA_VERSION_FULL-Windows-X64.exe"
makensis /V2 /P4 Ultimaker-Cura.nsi

usage: create_windows_installer.py [-h] source_path dist_path filename
create_windows_installer.py: error: unrecognized arguments: Ultimaker-Cura.exe 5 2 0 2 Ultimaker B.V. https://ultimaker.com ..\cura_inst\packaging\cura_license.txt LZMA ..\cura_inst\packaging\NSIS\cura_banner_nsis.bmp ..\cura_inst\packaging\icons\Cura.ico Ultimaker-Cura--Windows-X64.exe

jellespijker, 2022-10-25:
Am I correct in assuming that all the steps before the nsis installer ran without issues?

the NSIS installer step should still work, if it fails there you might want to look at your slashes in the command, maybe add quotes around them.

yess1987, 2022-10-31:
i do that.
But, how can I run、build、install the custom Cura/Uranium/CuraEngine? if I need to modify cura,Are there detailed steps or documents?

yess1987, 2022-11-01:
I modified part of cura code. Will the following command overwrite my changes.
“conan create . cura/5.3.0-alpha@yess/testing --build=missing --update”
Does the Build process use our locally modified code
"conan install cura/5.3.0-alpha@yess/testing --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst --json cura_inst/conan_install_info.json"

DerGenaue, 2022-11-01:
`conan create . yess/testing`
will take the current folder and package it into the local conan repository (somewhere in `C:\conan\...`).

`conan install cura/5.3.0-alpha@yess/testing --build=missing --update -g VirtualPythonRunEnv -o cura:devtools=True -if cura_inst`
will take the package out of the local conan repository and install it into the folder `cura_inst` wherever you run this command.

yess1987, 2022-11-02:
conan create . yess/testing
Trying to clone repo: https://github.com/ultimaker/conan-config.git
ERROR: Failed conan config install: Can't clone repo: Command 'git -c http.sslVerify=true clone "https://github.com/ultimaker/conan-config.git" .   ' returned non-zero exit status 128.
Cloning into '.'...
fatal: unable to access 'https://github.com/ultimaker/conan-config.git/': OpenSSL SSL_read: Connection was reset, errno 10054

When executing this command, do I have to update GitHub again, and can I only use the local cache

yess1987, 2022-11-03:
"conan install . cura/<version>@<user>/<channel> --build=missing --update"

 " cura/<version>@<user>/<channel>" ，must start with cura？

jellespijker, 2022-11-03:
@yess1987 can I ask where you're located? China perhaps?

On a separate note this has url has a trailing `/`
> fatal: unable to access 'https://github.com/ultimaker/conan-config.git/': OpenSSL SSL_read: Connection was reset, errno 10054

Please check your `conan config install https://github.com/ultimaker/conan-config.git` command and make sure it ends with `.git`

It would also help a lot with the readability of your comments if you use proper GitHub markdown style when writing see https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

yess1987, 2022-11-06:
Repeat the "create" and "install" commands several times, and the following commands appear.

ERROR: Invalid generator 'VirtualPythonRunEnv'. Available types: txt, gcc, compiler_args, cmake, cmake_multi, cmake_paths, cmake_find_package, cmake_find_package_multi, qmake, qbs, scons, visual_studio, visual_studio_multi, visual_studio_legacy, xcode, ycm, virtualenv, virtualenv_python, virtualbuildenv, virtualrunenv, boost-build, pkg_config, json, b2, premake, make, deploy, markdown, GitHubActionsBuildEnv, GitHubActionsRunEnv, PyCharmRunEnv, VirtualPythonEnv, CMakeToolchain, CMakeDeps, MSBuildToolchain, MesonToolchain, MSBuildDeps, QbsToolchain, msbuild, VirtualRunEnv, VirtualBuildEnv, AutotoolsDeps, AutotoolsToolchain, BazelDeps, BazelToolchain, PkgConfigDeps, VCVars, IntelCC, XcodeDeps, PremakeDeps, XcodeToolchain, MesonDeps

yess1987, 2022-11-07:
pyinstaller .\cura_inst\Ultimaker-Cura.spec

507 INFO: PyInstaller: 5.5
508 INFO: Python: 3.10.2
545 INFO: Platform: Windows-10-10.0.19044-SP0
551 INFO: UPX is not available.
575 WARNING: Unable to copy metadata for cura: The 'cura' distribution was not found and is required by the application
576 WARNING: collect_data_files - skipping data collection for module 'cura' as it is not a package.
577 WARNING: collect_dynamic_libs - skipping library collection for module 'cura' as it is not a package.
676 INFO: Determining a mapping of distributions to packages...
4344 WARNING: Unable to determine requirements for cura: The 'cura' distribution was not found and is required by the application
4359 WARNING: Unable to copy metadata for UM: The 'UM' distribution was not found and is required by the application
4360 WARNING: collect_data_files - skipping data collection for module 'UM' as it is not a package.
4360 WARNING: collect_dynamic_libs - skipping library collection for module 'UM' as it is not a package.
4479 WARNING: Unable to determine requirements for UM: The 'UM' distribution was not found and is required by the application
4494 WARNING: Unable to copy metadata for serial: The 'serial' distribution was not found and is required by the application
4495 WARNING: collect_data_files - skipping data collection for module 'serial' as it is not a package.
4496 WARNING: collect_dynamic_libs - skipping library collection for module 'serial' as it is not a package.
4599 WARNING: Unable to determine requirements for serial: The 'serial' distribution was not found and is required by the application
4614 WARNING: Unable to copy metadata for Charon: The 'Charon' distribution was not found and is required by the application
4614 WARNING: collect_data_files - skipping data collection for module 'Charon' as it is not a package.
4615 WARNING: collect_dynamic_libs - skipping library collection for module 'Charon' as it is not a package.
4718 WARNING: Unable to determine requirements for Charon: The 'Charon' distribution was not found and is required by the application
4735 WARNING: Unable to copy metadata for sqlite3: The 'sqlite3' distribution was not found and is required by the application
5414 WARNING: Unable to determine requirements for sqlite3: The 'sqlite3' distribution was not found and is required by the application
5428 WARNING: Unable to copy metadata for trimesh: The 'trimesh' distribution was not found and is required by the application
5429 WARNING: collect_data_files - skipping data collection for module 'trimesh' as it is not a package.
5429 WARNING: collect_dynamic_libs - skipping library collection for module 'trimesh' as it is not a package.
5544 WARNING: Unable to determine requirements for trimesh: The 'trimesh' distribution was not found and is required by the application
5560 WARNING: Unable to copy metadata for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
6272 WARNING: Unable to determine requirements for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
6286 WARNING: Unable to copy metadata for PyQt6: The 'PyQt6' distribution was not found and is required by the application
6287 WARNING: collect_data_files - skipping data collection for module 'PyQt6' as it is not a package.
6290 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6' as it is not a package.
6397 WARNING: Unable to determine requirements for PyQt6: The 'PyQt6' distribution was not found and is required by the application
6413 WARNING: Unable to copy metadata for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
6514 WARNING: collect_data_files - skipping data collection for module 'PyQt6.QtNetwork' as it is not a package.
6610 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6.QtNetwork' as it is not a package.
6812 WARNING: Unable to determine requirements for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
6828 WARNING: Unable to copy metadata for PyQt6.sip: The 'PyQt6.sip' distribution was not found and is required by the application
6927 WARNING: collect_data_files - skipping data collection for module 'PyQt6.sip' as it is not a package.
7023 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6.sip' as it is not a package.
7213 WARNING: Unable to determine requirements for PyQt6.sip: The 'PyQt6.sip' distribution was not found and is required by the application
7228 WARNING: Unable to copy metadata for stl: The 'stl' distribution was not found and is required by the application
7228 WARNING: collect_data_files - skipping data collection for module 'stl' as it is not a package.
7229 WARNING: collect_dynamic_libs - skipping library collection for module 'stl' as it is not a package.
7332 WARNING: Unable to determine requirements for stl: The 'stl' distribution was not found and is required by the application
7335 INFO: Extending PYTHONPATH with paths
['C:\\.conan\\675ef8\\1\\bin']
7862 INFO: checking Analysis
7921 INFO: Building because inputs changed
7922 INFO: Initializing module dependency graph...
7926 INFO: Caching module graph hooks...
7973 INFO: Analyzing base_library.zip ...
10195 INFO: Loading module hook 'hook-heapq.py' from 'C:\\Users\\shuan\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\PyInstaller\\hooks'...
[... cut ...]
15415 INFO: Analyzing hidden import 'PyQt6.QtNetwork'
15419 ERROR: Hidden import 'PyQt6.QtNetwork' not found
15420 INFO: Analyzing hidden import 'PyQt6.sip'
15420 ERROR: Hidden import 'PyQt6.sip' not found
15421 INFO: Analyzing hidden import 'logging.handlers'
15528 INFO: Analyzing hidden import 'zeroconf'
15529 ERROR: Hidden import 'zeroconf' not found
15530 INFO: Analyzing hidden import 'stl'
15534 ERROR: Hidden import 'stl' not found
15535 INFO: Analyzing hidden import 'serial'
15537 ERROR: Hidden import 'serial' not found

DerGenaue, 2022-11-11:
Here it was needed to allow people to beta-test the [TreeSupport 2 Fork](https://github.com/ThomasRahm/CuraEngine) (see the release section which basically contains "portable" zip files for Cura)

MariMakes, 2022-11-11:
Ahhhh, that makes sense! Super cool ⭐

closed 2022-11-26 by github-actions[bot]