Ultimaker/Cura #13569: Ultimaker-Cura-5.2.0-beta-win64 - VirusTotal (Cyren) reports W64/S-f7bcf467!Eldorado - most probably false positive

mm108, 2022-10-16:
### Application Version

Ultimaker Cura 5.2.0-beta

### Platform

Windows 10 Pro

### Printer

Ender 3

### Reproduction steps

VirusTotal (Cyren) reports W64/S-f7bcf467!Eldorado

This could pretty well be false positive. Screenshot attached. Thank you.

[image]

### Actual results

VirusTotal (Cyren) reports W64/S-f7bcf467!Eldorado

This could pretty well be false positive. Screenshot attached. Thank you.

### Expected results

VirusTotal (Cyren) reports W64/S-f7bcf467!Eldorado

This could pretty well be false positive. Screenshot attached. Thank you.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-10-17:
Hey @mm108,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for the report 👍 

We submit our files to be checked every release. It takes some time for all the virus checkers to accept our software. We haven't found a way to do this check before releasing without obstructing development. 

I recall that last time we were flagged, we changed our icon for a slightly different size but basically the same one, so I'm confident this is also a false positive. Usually, this resolves itself if we have enough users flagging it as a false positive. 

Are you blocked in any way to continue?

mm108, 2022-10-17:
hi @MariMakes,

Thank you for the reply. I was pretty sure too that it was a false positive but thought that I should bring it to your attention :)

The 5.1.1 release is working perfectly fine - it's only the 5.2.0-beta that shows this message. So it's all ok, no problem at the moment :)

Thanks once again. Cheers.

closed 2022-10-17 by MariMakes