Ultimaker/Cura #13572: Tree Support often only supports 2/3 of the needed area

Triangulix, 2022-10-16:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Slice the attached project

### Actual results

Tree support often misses 1/3 of the area that needs support.
[image]

### Expected results

The support should include the whole area that needs it.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Teeth_Cup.zip](https://github.com/Ultimaker/Cura/files/9795887/Teeth_Cup.zip)

GregValiant, 2022-10-17:
Thanks for the report.
From what I see in your setup the slice is going exactly according to your settings.  Two of those settings are the support overhang angle of 56° and a minimum support area of 5.0mm.  You are going to get a limited area of support with those numbers.
There are 73 settings in the Support Section.  You need to configure them to provide the support you want.
I'll leave this as a bug report for now but what I'm seeing is Expected Behavior given the settings you are using.

Triangulix, 2022-10-17:
Each of the marked areas have more than the set degree of overhang.
Do I understand you correctly that the option "Minimum support area" does not mean "support only areas of at least 5mm"?
As we both see the areas are supported but only partially.

MariMakes, 2022-10-17:
Hey @Triangulix,

When slicing your model and generating support, the slice engine examines every layer of your print. When support is generated, all areas smaller than the value for the Minimum Support Area will not be printed. I agree that it's not very intuitive, so we are looking at ways to improve this. 
https://support.ultimaker.com/hc/en-us/articles/6103470096028

GregValiant, 2022-10-17:
As for me, I'm just saying that given the correct settings that satisfactory support will be generated to hold up the overhangs while they print.
This is a nicely detailed model.  I would think that a .4 nozzle would give you a much nicer print because it would dictate a finer line width and that affects the support resolution as well as the model resolution.  You've chosen a .6 nozzle and that's fine but there will be trade-offs that need to be made since the support will be coarser.
[image]

[image]

[image]

jungletek, 2022-11-14:
> Thanks for the report. From what I see in your setup the slice is going exactly according to your settings. Two of those settings are the support overhang angle of 56° and a minimum support area of [...]

I don't know if it's a newer development, but that setting "Minimum Support Area" only applies to traditional supports, not tree supports; it is not an editable variable in 5.2.x while tree supports are selected.

GregValiant, 2022-11-14:
This is in the fdmprinter.def.json file and so all printers use it:
"enabled": "(support_enable or support_meshes_present) and support_structure == 'normal'"

So yes, it is not enabled when tree supports are selected.  The developers may have run into problems trying to implement it with tree supports.
Like some other settings, you my be able to switch to Normal supports to make the setting visible, adjust the setting, and then switch back to tree supports and Cura may use the new setting.  I don't know if that is true with this setting.

DerGenaue, 2023-07-04:
Please check if this has been solved with the new tree supports implementation in the latest version of Cura 5.4

MariMakes, 2023-07-11:
Hey @Triangulix ,

@DerGenaue is right, we've introduced a lot of improvements for tree supports in Cura 5.4.
I took a quick look and your model is wayyy better supported:
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

Triangulix, 2023-07-11:
@MariMakes You can close the issue. Maybe I'll try Cura 5.4 or a later version in the future.

closed 2023-07-11 by Triangulix