Ultimaker/Cura #13574: Unnecessary movement

Marceldu13, 2022-10-17:
### Application Version

5.2.0 Beta

### Platform

Mac OS 

### Printer

Sovol SV01

### Reproduction steps

When slicing my object , head is doing unnecessary movement leading to strings on my print check layer 10 11 .

### Actual results

Stringing on my print

### Expected results

Slicing leading to minimum travel to avoid stringing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[SS_Cul Bouteille V9.3mf.zip](https://github.com/Ultimaker/Cura/files/9799656/SS_Cul.Bouteille.V9.3mf.zip)
[profils.curaprofile.zip](https://github.com/Ultimaker/Cura/files/9799676/profils.curaprofile.zip)
[Materiaux.xml.fdm_material.zip](https://github.com/Ultimaker/Cura/files/9799678/Materiaux.xml.fdm_material.zip)
[Capture d’écran 2022-10-17 à 12.51.35.zip](https://github.com/Ultimaker/Cura/files/9799688/Capture.d.ecran.2022-10-17.a.12.51.35.zip)

GregValiant, 2022-10-17:
Thanks for the report.
5.x is different and there are more Wall settings than in the 4.x versions of Cura.  In this case your model is mostly walls and so close attention should be paid to those settings.
Try setting your "Wall Transitioning Threshold Angle" to 12° and see how it looks.  That was the only change I made and this was the result.  Those long combing moves are gone.
[image]

I'm going to remove the bug label and close this as it appears to be a settings issue.  If you feel that is premature let us know.  I can always put the bug label back on and reopen it.

closed 2022-10-17 by GregValiant

Marceldu13, 2022-10-17:
Thank you for such a quick and precise answer ! Will give it a try !

Eric Bloch
mobile: +33 6 09 48 97 04
email: ***@***.***
address: 131 Rue des Tappes 13680 Lançon-Provence

Pensez à l'environnement avant d'imprimer cet E-mail