Ultimaker/Cura #13578: Slice report error

JParr91, 2022-10-17:
### Application Version

5.1.1

### Platform

macOS Monterey 12.3.1

### Printer

Custom FFF

### Reproduction steps

Loaded model by clicking folder icon, model loaded, custom profile was selected, hit slice, got error.

### Actual results

error prompt indicated bug, refused to slice

### Expected results

model to be sliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[JPD Project 1.3mf.zip](https://github.com/Ultimaker/Cura/files/9803686/JPD.Project.1.3mf.zip)

I could not find the cura.log file as there was no cura folder in the application support library on my computer

MariMakes, 2022-10-18:
Hey @JParr91,

Thanks for your report 💪 

I was able to reproduce your error on my side. 
But I also noticed that by moving the model slightly in the Y direction I was able to slice it. 
[image]

Would this help you to get unstuck?

JParr91, 2022-10-18:
Yes that does help.
I also noticed that when I changed my support line thickness, that also fixed it. I change it from 0.6mm to either 0.5mm or 0.7mm and it was fine.
Thanks for the support!
JP

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker