Ultimaker/Cura #13581: Continuing problem with custom profiles

GoguyT3D, 2022-10-17:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

The custom profiles that I have saved in cura are not all showing, creating duplicates with #2 added to the end of the profile when I import the missing custom profile, showing all but 1 of my profiles are "not supported" when they were derived from supported generic profiles to begin with, and not the same ones are showing every time I startup cura. Also duplicate profiles show up as a different name which I have removed in the past.

### Actual results

I import the missing profile, it tells me the profile was imported as Name#2 meaning one already exists and it is making a duplicate although I can't see it. Restart cura and the profiles dont "stick" 

### Expected results

I Have 1 set of profiles backed up on my 3 drives, and would expect to be able to upload them once and have them stay visible all the time and not be labeled as "not supported" since they were made from modified default profiles to begin with.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I cant seem to get cura to maintain 1 set of profiles without it bringing back up old profiles and duplicates or hiding my current custom profiles.

MariMakes, 2022-10-18:
Hey @GoguyT3D,

Thanks for the report 💪 
Could it be that you changed your material type too much? 
That it's related to https://github.com/Ultimaker/Cura/issues/8229

For troubleshooting, I need your configuration folder. 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

GoguyT3D, 2022-10-18:
So I have tried one thing since I submitted this post: After reading article #8229 from top to bottom I am beginning to realize why cura is doing this. Since I posted this bug I have gone into my custom materials for my 3rd party filaments and changed the material type back to what it started as(for instance my PETG profile I changed back to CPE) because all my materials did start as a default generic material I just adjusted settings to fit my filament. Since doing that my custom profiles re-appeared and are visible yet still "not supported"
[themes.zip](https://github.com/Ultimaker/Cura/files/9815211/themes.zip)
My config folders are added

GoguyT3D, 2022-10-18:
So it is still unclear to me why a generic material such as CPE which is supported by the U3S is labeled as not supported, is that just because its not supported via the warrantee?

MariMakes, 2022-10-19:
Hey @GoguyT3D,

Thanks for sharing the extra files. I think I understand what is happening. 
There are two things to keep in mind with the not supported screen.

1. It pops up if there is an error with a Dual extrusion combination or a material - print core combination. That means that the first thing to check is if your cores are defined correctly and if you want to print single extrusion that your other core should be disabled.
2. When making custom materials it's super important that you match the Material Type exactly to have the resolution show up correctly. I've noticed thta your TPU 95A was defined as TPU instead and this was tripping up Cura. If you restore it to 95A you should see the correct resolutions.

https://user-images.githubusercontent.com/40423138/196685041-530b2b5b-0b33-4d09-81c2-08aa860c3d80.mp4

Does this help you get unstuck or did I look at the wrong thing?

GoguyT3D, 2022-10-20:
I have attached a photo of the changes you suggested I make. Considering what you said about printer core 2 being disabled, sometimes I use my second print core with my custom profiles and sometimes I dont. I'm sure that back then I was unaware of whether the nozzle 2 was disabled or not when I most recently updated my profiles although in the future I will be more diligent regarding all of the settings or cores which are enabled when saving/exporting my custom profiles to make sure there is no problems in the future. I have also been having problems syncing these profiles to my printer but I think I have another ticket for that open already. Thank you for the help on this ticket, I will be more focused on all settings including core, material type, profile and names.
[image]

closed 2022-10-20 by GoguyT3D