Ultimaker/Cura #13582: Slicing an STL file results in an error

herrbrave, 2022-10-17:
### Application Version

5.2.0-beta

### Platform

Windows 11

### Printer

Creatlity Ender 5 Pro

### Reproduction steps

1) load file
2) slice file

### Actual results

Shows an error message with a request to report a bug.

### Expected results

File sliced properly.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[stderr.log](https://github.com/Ultimaker/Cura/files/9804413/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/9804414/stdout.log)
![image](https://user-images.githubusercontent.com/168682
[cura.log](https://github.com/Ultimaker/Cura/files/9804496/cura.log)
14/196282059-ef30749f-5977-43ca-b0dc-29611441c9dc.png)

MariMakes, 2022-10-18:
Hey @herrbrave,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report, but to help you we need your file and settings. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

MariMakes, 2022-10-18:
Hey @herrbrave,

Thanks for sharing your .stl file but that is not what I'm looking for. 
If I load the STL into Cura and I select my Ultimaker printer, this model slices normally so there is a combination of printer + settings that is causing a crash. 

Could you please share a project file/.3mf instead?. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

herrbrave, 2022-10-18:
[CE5P_Mansion_Ground_Floor_B.zip](https://github.com/Ultimaker/Cura/files/9812375/CE5P_Mansion_Ground_Floor_B.zip)

MariMakes, 2022-10-19:
Hey @herrbrave,

Thanks for sharing the file 😊 

I can recreate the crash on our side so I'll bring it up to the team.

It might be a rounding issue because when I restore the layerheight from 0.32 to 0.2 I am able to slice. 
I hope this helps you get unstuck. 
[image]

herrbrave, 2022-10-20:
Great, I'm glad you were able to repro the issue. I was able to slice the file with the same config in Cura 5.0 successfully to unblock myself. Thank you for looking into it.