Ultimaker/Cura #13584: Pause at layer height

ydkjman, 2022-10-17:
### Application Version

5.1

### Platform

Win 10 Pro

### Printer

Ender 5 Pro

### Reproduction steps

Pause at height (I used layer)

### Actual results

Every thing works up until after you have changed out your filament.  once you resume print it reaches temp it used to extrude a small amount of filament out  before it would move over the print and start printing.  In 5.1 is now moves over the print and then extrude the amount on top of the print.

### Expected results

it usually would extrude before it would move over the print and start printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-10-17:
Thanks for the report.
This might be a duplicate of #11471.  To be sure, could you post a gcode file that had the problem?  You will have to zip it and post the zip folder here.
It appears that Cura is indeed moving the print head over the model and then doing a prime using the retraction number even when no prime is called for in the Pause setup.
M0 ; Do the actual pause
M109 S210 ; resume temperature
G1 F300 Z1.2
G1 F9000 X87.29 Y122.017
G1 F300 Z1.2 ; move back down to resume height
**G1 F1500 E25**

When no "extrude" amount was entered I think it should be:
M0 ; Do the actual pause
M109 S210 ; resume temperature
G1 F300 Z1.2
G1 F9000 X87.29 Y122.017
G1 F300 Z1.2 ; move back down to resume height

and if an extrude was requested then it should be prior to moving back over the print:
M0 ; Do the actual pause
M109 S210 ; resume temperature
**G1 F1500 E25**
G1 F300 Z1.2
G1 F9000 X87.29 Y122.017
G1 F300 Z1.2 ; move back down to resume height

ydkjman, 2022-11-01:
Guess what:  5.2.1 still does the very same thing when doing a pause at height and setting it to a layer.  When you hit resume, it moves over the print and extrudes the bit of filament on the print leaving a big blob on the print for the nozzle to hit and knock the print right off the plate.

Again, it used to extrude the small amount before it would  move over and resume printing.

What am I doing wrong

GregValiant, 2022-11-01:
I think it's a bug in the Pause at Height script and the "prime" command needs to be moved.  I've re-written it and am trying to submit a pull request.   When I figure that out then we'll see.

ydkjman, 2022-11-01:
> I think it's a bug in the Pause at Height script and the "prime" command needs to be moved. I've re-written it and am trying to submit a pull request. When I figure that out then we'll see.

Awesome thanks so much.