Ultimaker/Cura #13585: Unexpected error

raul7321, 2022-10-17:
### Application Version

5.1.1.

### Platform

Windows 10

### Printer

Prusa i3

### Reproduction steps

Press "Slice" button

### Actual results

An unexpected error

### Expected results

The result of slice process

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[PI3_Dual.zip](https://github.com/Ultimaker/Cura/files/9804993/PI3_Dual.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9805000/cura.log)

MariMakes, 2022-10-18:
Hey @raul7321,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report. I can reproduce your issue in Cura 5.2.

I'll have someone from the Cura team take a look at the crash. 

I noticed that if you select a different layerheight, you can slice the model.
I hope that helps you to get unstuck.
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker