Ultimaker/Cura #13590: 5.1.1 View manipulation not working: zoom/rotate/panning

Distillah2, 2022-10-18:
### Application Version

Cura 5.0.0 and .1.1

### Platform

Win 10

### Printer

Creality Ender 6

### Reproduction steps

When Cura opens and when a model is inserted and after slicing, none of the mouse buttons for zooming, panning and rotating are working. None of the view buttons are working either (lower left corner).
This started in Cura 5.0.0 and is still happening in the update to 5.1.1.

### Actual results

None of the view buttons on mouse or lower left corner are working.
The screen as shown stays in that position, what ever viewing command is given
[image]

### Expected results

The view should follow viewing instructions

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No further additional information

timmackey0851, 2022-10-27:
Ive also reported the same issue

MariMakes, 2022-11-08:
Hey @Distillah2,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13590
We have quite a number of reports about this, so I'll be bringing it up with the team. 
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13590

closed 2022-11-08 by MariMakes