Ultimaker/Cura #13591: slicer: out of order thin wall slicing regression for planeprint.com

bassco, 2022-10-18:
### Application Version

5.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Attached an STL And the PDF to setup Profile 3 for Thin Wall printing from [planeprint.com](htpps://planeprint.com/print)
2. Create a Profile 3 profile
3. Slice the WingTest file

Preview the last layer. You will see that the inner wall is printed before the outer wall. Both walls are treated as "OUTER WALL" in the sliced gcode.
[planeprint.zip](https://github.com/Ultimaker/Cura/files/9812276/planeprint.zip)

### Actual results

The top 4 layer have an extra wall layer that should print after the outer layer. However, since 5.x, this is not the case.

[image]
[image]

### Expected results

With Cura 4.12 it can be seen that the outer wall is being printed on the existing layers and therefore, the inner wall has a surface to grip onto.

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I have also exported the Cura 4.12 and Cura 5.1.0  perojects with the STL and settings (planeprint-projects.zip)

[planeprint-projects.zip](https://github.com/Ultimaker/Cura/files/9812346/planeprint-projects.zip)

The 5.2 release exhibits the slicing difference as 5.1

GregValiant, 2022-10-18:
Thanks for the report.
Could you post a Project File?  Use the File | Save Project command and zip the 3mf file and post the zip file here.  It would be much easier to troubleshoot than going over the 4 pages of instructions.
The only way I've gotten it to slice correctly is using "surface" mode.
When a wall has air on one side and model on the other it is an outer wall.  So that top edge that faces inward is by definition another outer wall.  If I had designed that I would have put a chamfer on that second wall so it wouldn't need support. because yes, it appears to be printing on air.  Knowing how you've set this up (Project File) would go a long ways.

bassco, 2022-10-19:
Hi, @GregValiant 

I included project files for Cura4.12 and Cura5.1 in the original post. 

PlanePrint-projects.zip in the Additional Comments block.

Surface mode is enabled too.

GregValiant, 2022-10-19:
Sorry.  I opened the first file last night and it was just the STL.
FYI I've built a lot of U-Control planes and a couple of RC planes and one RC boat.  Printing planes and boat hulls comes up quite often on the UM Forum.  Planes are tough because nobody sticks an OS-90 into them anymore and so the weight is really important.  Too bad really.  I always liked the smell of burning nitro-methane and castroil.

The double surfaces at the top and bottom are "parallel" everywhere and are 0.40mm apart.  No settings in Cura can change that as the model is the model.
A consequence is that when in surface mode - no adjustment can be made for "where" the surfaces are.  Cura puts plastic down right on the wall.  So at 0.4 line width the edges of the extrusions barely touch and at the top edge, the extrusion falls to the build plate as it doesn't have any support.  You have found this out.

Here I've set the line width to 0.3 to highlight the problem (these images are from 4.12.1).
[image]

Here is just the top layer at .5 line width.
[image]

Printing in Surface Mode takes a lot of settings out of consideration.  I thought I could make a change with a Support Blocker configured as a Cutting Mesh.  It worked in 4.12.1 but not in 5.2.  In 5.2 the Cutting Mesh will accept the Outer Wall Line Width settings, but it doesn't change anything.  In 4.12.1 it does make a difference.  That looks like another bug to me.

So it would appear that because both walls are "Outer Walls" that there is no way to explain to Cura to print "This outer wall first.".  That, coupled with the .4mm overhang of the inner wall leaves it unsupported if Cura decides to print it first.  Cura 4.12.1 appears to done that but 5.2 seems to want to do it the other way and it doesn't work at 0.4 line width.

I sank the model 50mm into the build plate and with the "Outer Wall Line Width" at 0.50 it printed very well.  I know this makes a slight difference in the width of the finished part as well as the weight (5.4gr at .4 line width and 6.4gr at .5 line width).

I'll leave this as a bug report but it may be better as a feature request.  Cura is working as it was designed, but it doesn't work so well for what you need to do.
I think the inability of the Cutting Mesh to make a difference qualifies as a bug in it's own right.

bassco, 2022-10-19:
Love me the smell of nitro in the morning!

I suspect it has something to do with the change to the arachne slicer as the 4.x Cura series was flawless and the behaviour change showed up in the 5.x versions. I have not tried 5.0.

I'll continue to use 4.12 for these designs as you suggest. 

Thanks for investigating @GregValiant 

Adding the physical print that shows the outer of order printing in 5.1. Sometimes easier to see physical things.

[image]

bassco, 2022-10-19:
> Sorry. I opened the first file last night and it was just the STL.

Yeah, I realised after reading through the issue template that a project file would be more useful - you were really quick picking this ticket up. I must have edited the upload after you grabbed it.

GregValiant, 2022-10-19:
Yeah, those strings are from the inner not sticking and dragging across the curves instead of following them.
A couple of things in your settings that could apply:

Your settings are:
- Print Temperature 230°.
- 60mm.sec print speed
- Cooling at 60%.
- Outer Line Width at 0.4mm
You are extruding very hot, at a fair clip, and not cooling as much as you could, and your line width isn't sufficient to touch the other wall (even if it was there).

My settings would be:
- Print Temperature 205°
- 35mm/sec print speed
- Cooling at 100% (single 5015 blower)
- Outer Wall Width 0.5mm

That inner wall printed first but stuck.  The plastic cooled quickly enough to stay where it was put.  After the first layer was there the others had no problem.  (I printed on a brim since there is only the single wall on this partial.  Brim distance was 0.1 and the brim came right off.)
[image]
[image]

bassco, 2022-10-20:
The PLA is PolyAir 1.0 from 3DLabPrint.com and has a 220-240 range for good layer adhesion and excellent impact resistance, as demonstrated by turning the sample print inside out and not having any layer separation.

[image]

I followed the recommended part cooling of 20% fan speed, but I was getting cooling issues and bumped it to 60%, which gave better results. Print speed of 60mm/s was dropped from 80-100mm/s, as I initially thought the speed was the bonding issue I was seeing. It was only later when I viewed the Preview, that I saw the changed slicer behaviour. 

I have a fixed bed CoreXY printer that is running klipper and input shaper and produces really nice prints at higher speeds.

I've tested 4.12 and it slicers in the correct order, so I will fallback to using that version for now. 

Thanks for following up and your 0.50mm wall thickness idea.

Should I create a ticket on CuraEngine and link it to this discussion?

GregValiant, 2022-10-21:
_"Should I create a ticket on CuraEngine and link it to this discussion?"_
That's up to you.  I open bug reports when I find a problem but I like to provide a full description of the problem.  I don't know the background code but I'm sure it's complicated and the more information you can give, the better chance the Cura developers can find the specific area in the code that needs attention.  Make sure you mention that the behavior in Surface Mode was different in the 4.x versions.  (I really liked 4.13.1 by the way.)

In this case one of the problems is in the model where that inner surface "appears" up high.  It's a .4 gap and as we discussed - at 0.4 line width the extrusions don't quite touch but at 0.5 line width there is some overlap.  When I have a situation like that on my own models I give it a chamfer so it builds going outward instead of just "appearing".

I've taken to printing Vase Mode and Surface Mode models at .6 to .8 line width.  The walls are much more robust but for airplane parts there is the weight issue and a wider wall width isn't always an option.
The "Print Order" is a definite problem and you shouldn't have to compromise with a solution that makes the model heavier.  It would be nice to be able to dictate the print order.  I'd call that a "Feature Request".

It would also be nice if more settings worked when using Infill Meshes in Surface Mode.  Right now you can configure a blocker as an Infill Mesh with certain settings, but they don't appear to do anything.  That looks like a "Bug" to me.

bassco, 2022-10-21:
@GregValiant - I've looked through the issues in CuraEngine regarding Surface Mode and I think this is a good use case for them to investigate.

Thanks for the ideas to enable a good quality print. I feel it's a workaround for functionality that was available prior to the change to the arachne slicer algorithm and should be addressed there.

Many thanks for your feedback and interest to solve this.